FreeCAD/FreeCAD #12336: [Problem] AddonManager cannot handle comments in README markdown

pieterhijma, 2024-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Addon Manager shows the README files of addons. However, the Addon Manager cannot handle comments in Markdown in any way known to me. Unfortunately, Markdown has no standard syntax for comments, so it is always unclear what a Markdown parser supports.

Why this is important: My project attempts to be [REUSE compliant](https://reuse.software/) which requires to have SPDX license information in each file (and each of the used licenses in directory `LICENSES`). All files including `README.md` are taken into account and therefore, I would like to add the SPDX information into the README in a comment.

Initially, the SPDX license information was included in an HTML-style comment like ``, but this results in a garbled view in the Addon Manager where the images are shown and only some bullets. Removing the HTML comments results in the expected behavior, so apparently HTML comments are not allowed by the parser.

To handle this situation, I experimented with the following suggestions obtained from [Stack overflow](https://stackoverflow.com/questions/4823468/comments-in-markdown) and [W3Schools.io](https://www.w3schools.io/file/markdown-comments/), but none are working and all are shown verbatim in the Addon Manager:

- `[//] # (SPDX-License-Identifier: LGPL-3.0-or-later)`,
- `[comment] <> (SPDX-License-Identifier: LGPL-3.0-or-later)`,
- `[SPDX-License-Identifier: LGPL-3.0-or-later]::`, and:

``` markdown
{::comment}

SPDX-License-Identifier: LGPL-3.0-or-later

{:/comment}
```

Unless anyone has a proper way to add a comment to Markdown that Qt accepts, I would propose to add a `cleanMarkdown` method that removes HTML comments from the Markdown text. In fact, because of experimentint with this problem, I could immediately submit a PR for this if this is the preferred solution for this problem.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35859 (Git)
Build type: Unknown
Branch: addon-manager-gitea-markdown
Hash: 0cdf43925de27f306ba211723b0309432d7ef4b2
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * backup 0.2.2
  * BIM 2021.12.0
  * Assembly3 0.12.2
  * fasteners 0.5.0
  * Assembly4 0.50.6
  * osh-autodoc-workbench 0.2.2
  * DynamicData 2.60.0
```
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-02-10:
Your solution of supporting HTML-style comments is fine with me -- if you have a PR ready to go, let's take that route.

chennes, 2024-02-11:
(Of course, another solution is to simply allow the license information to be visible in the rendered text, and not try to hide it.)

pieterhijma, 2024-02-14:
That would be so pragmatic... :smile:. I tried it but then my email address is rendered on an HTML page and I would like to think that I have at least some control over where to have it shown up :smile:.

pieterhijma, 2024-02-14:
The solution to this problem is simple. After some investigation, it turns out that Qt supports GitHub flavored Markdown by default. GitHub flavored Markdown should support [HTML-style comments](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#hiding-content-with-comments). It turns out that the comment syntax is only supported on a single line and not over multiple lines.

So, for anyone wanting to hide things in READMEs for the Addon Manager, you can add comments by using HTML-style comments ``.

closed 2024-02-15 by chennes