FreeCAD/FreeCAD #12343: [Problem] Undo/Redo for sketcher constraints don't work correctly

wwmayer, 2024-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch
2. Create e.g. a rectangle
3. Select a line and set a length constraint
4. Select the length constraint and move
5. Click on undo
Result: the length constraint disappears and the redo button is greyed out

Expected result: the constraint should be moved to the old position and the redo button should become active

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35859 +2 (Git)
Build type: debug
Branch: fix_port_occ_780
Hash: b129029745cff523bcee87b2abe85a550e0a0da8
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-02-11:
@PaddleStroke and/or @AjinkyaDahale can one of you guys take a look?

PaddleStroke, 2024-02-12:
So if I understand correctly you suggest that the move constraint is done inside a transaction correct? Was this already the case and got broken or is it a feature request? Just to know what I'm looking for.

AjinkyaDahale, 2024-02-12:
@PaddleStroke the redo button going away is more concerning. Apart from that, ideally since we are providing the ability to move the dimensions around they should stay there and be undoable.

wwmayer, 2024-02-12:
> So if I understand correctly you suggest that the move constraint is done inside a transaction correct?

I don't mind. It's probably good to add a transaction because when moving a geometry also a transaction is opened.

But this is not my actual point. The bug is that if you press the undo button after moving a constraint it will be removed and it's not possible any more to restore it because the redo button is disabled.

If you create and delete a constraint without having moved it the undo/redo works as expected.

PaddleStroke, 2024-02-12:
Ok I see, the 'Drag constraint' is not on the undo stack like it should. So doing undo undoes the previous action. At that point the 'Drag constraint' appears which breaks the redo.
- This behavior is already like this in 0.20 so it's not new.
- Drag curve works as expected
- Pressing recompute makes the 'drag constraint' appears in the undo stack. But undoing it doesn't work.

PaddleStroke, 2024-02-12:
Ah so comparing drag curve and drag constraint I see that drag curve is doing a                                 `tryAutoRecomputeIfNotSolve(getSketchObject());`. Adding that to drag curve does solve the problem as the drag constraint appears in the stack.

Now undoing this transaction still does nothing. I think that this is because it is only undoing the very last constraint move that happens on the release. The initial position doesn't seem to be registered anywhere.

PaddleStroke, 2024-02-12:
I do not know well the transaction system. Can you please advise : Is that suppose to work?

```
                        getDocument()->openCommand(QT_TRANSLATE_NOOP("Command", "Drag Constraint"));
                        auto idset = drag.DragConstraintSet;
                        for (int id : idset) {
                            moveConstraint(id, Base::Vector2d(x, y));
                        }
                        getDocument()->commitCommand();
                        tryAutoRecomputeIfNotSolve(getSketchObject());
```

Or does the command 'actions' has to be 'registered' by wrapping them in 'cmdAppObjectArgs' ? Like drag curve : 

```
        getDocument()->openCommand(QT_TRANSLATE_NOOP("Command", "Drag Curve"));
        Gui::cmdAppObjectArgs(getObject(),
                     "movePoint(%d,%d,App.Vector(%f,%f,0),%d)",
                       drag.DragCurve,
                       static_cast<int>(Sketcher::PointPos::none),
                       vec.x, vec.y, drag.relative ? 1 : 0);
       getDocument()->commitCommand();
       tryAutoRecomputeIfNotSolve(getSketchObject());
```

PaddleStroke, 2024-02-12:
If we need to use cmdAppObjectArgs, then I guess we need to add a python binding to the moveConstraint function right? And VPSketch does not have a ViewProviderSketchPy.cpp so I guess this needs to be implemented too?
Let me know thanks

wwmayer, 2024-02-12:
> This behavior is already like this in 0.20 so it's not new.

I don't have a v0.20 but I can confirm the bug in v0.21.

> Pressing recompute makes the 'drag constraint' appears in the undo stack. But undoing it doesn't work.

Very strange behaviour. The command shows up in the undo stack but undoing it has no effect. And if a press the Recompute button before dragging the constraint then the command doesn't show up in the undo stack.

> Can you please advise : Is that suppose to work?

Yes, it should work like that.

> Or does the command 'actions' has to be 'registered' by wrapping them in 'cmdAppObjectArgs' ? Like drag curve :

No, that's not needed. The relevant point is that properties of objects are modified. The document then stores the old values in the transaction objects.

> If we need to use cmdAppObjectArgs, then I guess we need to add a python binding to the moveConstraint function right? And VPSketch does not have a ViewProviderSketchPy.cpp so I guess this needs to be implemented too?
Let me know thanks

For a correctly working undo/redo all this is not needed.

PaddleStroke, 2024-02-12:
Well this is strange then. Because moveConstraint is called at 2 places : 
- In mouseMove
- In mouseButtonPressed when button is released.

The first thing is that the 'DragConstraint' command is created and commited only in mouseButtonPressed. So the first thing is that undoing it would undo the last call to moveConstraint. So it would virtually be invisible as the constraint moved in mouseMove just before the button release.

However I have just commented out the moveConstraint in mouseMove to test. So now the constraint does not move until you release the button. So now the constraint movement is all done during the transaction.
But undo is still not working.

wwmayer, 2024-02-22:
I have added #12548 to access the LabelDistance and LabelPosition members of the Constraint class via Python.
The code snippet below clearly shows that iundo/redo basically works:
```
obj=App.ActiveDocument.ActiveObject
obj.getLabelDistance(0)
doc=App.ActiveDocument
doc.openTransaction("Move label")
obj.setLabelDistance(0, 10)
doc.commitTransaction()
```

To test the code create a sketch with a single line. Add a length constraint and make sure that here isn't another constraint defined. Don't leave the edit mode and execute the code. Now you can press undo & redo and see that the label switches.

wwmayer, 2024-02-22:
I have discovered the actual problem:
The function moveConstraint() is called for every mouse move event and internally it directly changes the LabelDistance of the constraint.
When releasing the mouse mouseButtonPressed() is called that opens the transaction and modifies the constraint. But the LabelDistance is already set to the latest value and thus the undo/redo doesn't show any effect.

PaddleStroke, 2024-02-22:
Yes I noticed that (see above posts). But even commenting out the movement
during mouse Move, the undo was not working correctly.

wwmayer, 2024-02-22:
https://github.com/FreeCAD/FreeCAD/pull/12554

commit a38e7313 (2024-02-24): Sketcher: fixes #12343: Undo/Redo for sketcher constraints doesn't work correctly
closed 2024-02-24 by wwmayer