FreeCAD/FreeCAD #12345: [Problem] Sketcher Construction Lines are displayed in the incorrect locations on certain distros

Syres916, 2024-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A picture tells a thousand words so:

[image]

The handles are in the correct locations but the line appears to be mirrored about the X axis. A Git Bisect on the Fedora 38 system determined the commit https://github.com/FreeCAD/FreeCAD/commit/b140fea to be the root cause. This bug can be seen on both Fedora 38 and Debian 12.x but is not present on the Conda AppImage and Linux Mint 20.3 self compiled.

See forum discussion https://forum.freecad.org/viewtopic.php?t=85137

So for me the question is, if this is down to library dependencies (Coin) then is the community no longer supporting these two distros until they are brought to the same version level or should https://github.com/FreeCAD/FreeCAD/pull/11996 be reverted for the time being? It's not like dotted lines are essential for sketching but trying to make two parallel construction lines as it stands is a non-starter.

### Full version info

```shell
Bug present using:

OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35868 (Git)
Build type: Release
Branch: main
Hash: ad2fb73bffeb8d6d6739328379a4571a88834449
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.24.8
  * Assembly3 0.12.0
  * Plot 2022.4.17
  * freecad.gears 1.1.0
```

```
OS: Fedora Linux 38 (Thirty Eight) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35858 (Git)
Build type: Release
Branch: main
Hash: 3a73f916ddab6a0af75265acef7cc87e44e4664a
Python 3.11.7, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.0
  * sheetmetal 0.3.15
```

Works as expected using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35858 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a73f916ddab6a0af75265acef7cc87e44e4664a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.5
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35863 (Git)
Build type: Release
Branch: main
Hash: 4988894a924a1e6528bcdfbbb05211ab309908ea
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CurvedShapes 1.0.5
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-02-11:
Here is a bit of Python that also demonstrates the problem:

```
import FreeCADGui as Gui
import FreeCAD as App
from PySide import QtGui
from PySide import QtCore

from pivy import coin

#setup the contour
color = coin.SoBaseColor()
color.rgb = (0, 0, 1)

points = coin.SoCoordinate3()
lines = coin.SoLineSet()

boundary_style = coin.SoDrawStyle()
boundary_style.style = coin.SoDrawStyle.LINES
boundary_style.lineWidth = 3
boundary_style.linePattern = 0xE4E4
boundary_style.linePatternScaleFactor = 5

points.point.values = ( (0,0,0),(10,10,10),(10,10,0) )

#feed data to separator
sep = coin.SoSeparator()
sep.addChild(points)
sep.addChild(boundary_style)
sep.addChild(color)
sep.addChild(lines)

#add separator to sceneGraph
sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
sg.addChild(sep)

Gui.ActiveDocument.ActiveView.viewFront()
Gui.ActiveDocument.ActiveView.fitAll()

```

The result should look like  `/|` but on Fedora 38 and Debian 12 it appears as `\|`

adrianinsaval, 2024-02-13:
@PaddleStroke have you got any clue on why this might be happening?

PaddleStroke, 2024-02-14:
From the OP test where he use coin directly to create lines, it seems likely to be a Qt bug. I asked the OP to perform some more tests but he refused and I don't have a Fedora38 or Debian12 machine to test so it is hard to know fore sure.

But it seems that the bug comes from the use of `boundary_style.linePattern = 0xE4E4` that induce this bug. Someone who can reproduce the bug need to test the above code without the linePattern modification to see if the line draws correctly.
If it is confirmed, then we need to put a distro detection in the code to set Fedora/Debian users lines pattern to only use normal lines.

luzpaz, 2024-08-24:
bumping for visibility

TheYang, 2024-09-09:
I have this issue as well in my Debian12 VM in which I sometimes work.  

trying the Python code from [here](https://github.com/FreeCAD/FreeCAD/issues/12345#issuecomment-1937890674) results in:  
[image]

after commenting out "boundary_style.linePattern = 0xE4E4" it results in:  
[image]

So it seems the line pattern is the issue

closed 2025-07-07 by github-actions[bot]