FreeCAD/FreeCAD #12348: issue in non-linear material FEM

MatheoBARRIER, 2024-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello , I'm a student and I'm working on FreeCAD,
I have a problem on the FEM part of FreeCAD: 

To enter the non-linear properties of the material, we have to enter each points of the curve.
[image]

But the data I get doesn't match with the initial curve. (blue curve: datas entered in FreeCAD, yellow curve: FreeCAD calculations)
[image]

I choose the "von Mises" stress to obtain those datas, the other strains like "Major principal Stress" doesn't work either.

I tested several materials like polypropylene, ABS, Onyx and the stress is always approximately 10MPa lower than expected.
I noticed that the Young modulus as an influence on the constraint: if it's < 5000MPa the calculated stresses are much lower than they should be.

So I'm active in this project and I'm still waiting for response.

### Full version info

```shell
** written by FreeCAD inp file writer for CalculiX,Abaqus meshes
** highest dimension mesh elements only.

** Nodes
*Node, NSET=Nall
1, 4, -10, -54.13036777966
2, 0, -10, -54.13036777966
3, 4, -5, -39.46748948104
4, 0, -5, -39.46748948104
5, 4, -5, 40.53251051896
6, 0, -5, 40.53251051896
7, 4, -10, 55.19538881757
8, 0, -10, 55.19538881757
9, 4, -10, 85.54538881757
10, 0, -10, 85.54538881757
11, 4, 10, 85.54538881757
12, 0, 10, 85.54538881757
13, 4, 10, 55.19538881757
14, 0, 10, 55.19538881757
15, 4, 5, 40.53251051896
16, 0, 5, 40.53251051896
17, 4, 5, -39.46748948104
18, 0, 5, -39.46748948104
19, 4, 10, -54.13036777966
20, 0, 10, -54.13036777966
21, 4, 10, -84.48036777966
22, 0, 10, -84.48036777966
23, 4, -10, -84.48036777966
24, 0, -10, -84.48036777966
25, 4, -5.466, -84.48036777966
26, 4, 0.494, -84.48036777966
27, 4, 5.864, -84.48036777966
28, 4, 10, -79.39067277966
29, 4, 10, -72.02472777966
30, 4, 10, -64.34921277966
31, 4, 10, -58.62216777966
32, 4, 5.509952226548, -44.38863320999
33, 4, 7.426067912154, -49.98250038303
34, 4, 5, -34.85148948104
35, 4, 5, -28.91548948104
36, 4, 5, -20.97948948104
37, 4, 5, -12.65148948104
38, 4, 5, -4.315489481043
39, 4, 5, 4.012510518957
40, 4, 5, 12.34851051896
[... 1656 lines cut ...]
**  TimeSpan: s
**  Temperature: K
**
**  This leads to:
**  Force: N
**  Pressure: N/mm^2 == MPa (Young's Modulus has unit Pressure)
**  Density: t/mm^3
**  Gravity: mm/s^2
**  Thermal conductivity: t*mm/K/s^3 == as W/m/K == kW/mm/K
**  Specific Heat: mm^2/s^2/K = J/kg/K == kJ/t/K
**
```

### Subproject(s) affected?

FEM

### Anything else?

this is the material properties I'm using.
[materials.zip](https://github.com/FreeCAD/FreeCAD/files/14236549/materials.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-12:
Keep in mind that when you define plasticity in FreeCAD FEM, you have to enter true stress vs true plastic strain. Your data looks like total strain. The first point should be yield stress vs zero plastic strain.

MatheoBARRIER, 2024-02-19:
Sure, I entered true stress vs true plastic strain because the data comes from a tensile real tensile test.
But I don't understood, you want me to modify the yield stress to match with the curve ?

Anyway, do you know why the Young modulus as an impact on results? According to my research, the Young modulus is not in any constraint calculations.

FEA-eng, 2024-02-19:
Ok, if the data is true ([not engineering](https://enterfea.com/true-stress-true-strain/)) stress/strain then fine. But in the CalculiX syntax, the first point should be yield stress vs 0 plastic strain. So the specified curve describes only the plastic part (elastic one is defined separately with Young's modulus and Poisson's ratio). After this first point, you specify further stress-plastic strain pairs to define the whole curve.

It's also important how you read the results. You should select a representative element and node and use proper stress/strain components (e.g. S33 and E33 if tension is in Z direction). Or use total force and displacement outputs and calculate stress and strain from them. The concept of true vs engineering stress/strain applies here again. This CalculiX example can be a good starting point: https://github.com/calculix/CalculiX-Examples/tree/master/NonLinear/TensileTest

MatheoBARRIER, 2024-02-19:
I tried to change the yield stress, even if I enter 0 there is no changes.

Btw, I read the results by doing a tensile test on a tensile specimen, I'm entering the displacement and I read the Von mises stress on CalculiX. The "major principal stress" and the "stress zz component" on the pipeline have similar results.
[image]

If I doesn't understood, can you show me an example of changes? this is the curve points:

4.9,0.003
10,0.007
15.4,0.014
18.4,0.021
21.1,0.029
25.2,0.046
27.1,0.064
27.9,0.087
28.2,0.104
28.3,0.121

The polypropylene I'm using has a Young modulus at 1432 MPa and a Yield stress at 25.94 MPa.

Thanks
.

FEA-eng, 2024-02-19:
> Yield stress at 25.94 MPa

Then this should be the first specified point:

`25.94, 0`

After that, you should add the remaining points with non-zero plastic strain. This would mean that your strains are total strains. In such a case, you should convert them to plastic strains: `ε_plastic=ε_total-(σ/E)` where both `ε` and `σ` are true (not engineering) measures. This blog post explains it nicely (check especially "Important note 2"): https://simulation-blog.technia.com/simulation/converting-engineering-stress-strain-to-true-stress-strain-in-abaqus

MatheoBARRIER, 2024-02-20:
The material I want to study has a non-linear elastic curve, if I understand properly you want to make the software understand that I'm entering a plastic curve.

You want me to set the Yield stress at the first point of the curve but for me, the yield stress is at the end:
[image]

For information, I calculate the true strain : ln(1 + engineering strain)
                                                  true stress: exp(2 * poisson ratio * true strain)

FEA-eng, 2024-02-20:
> The material I want to study has a non-linear elastic curve,

I see, that's important and common for plastics (polymeric materials). But then you should use nonlinear elasticity (hyperelasticity) which requires manual CalculiX keyword edits or the use of Elmer [Deformation equation](https://wiki.freecad.org/FEM_EquationDeformation).

Assuming linear elasticity could be highly inaccurate here.

MatheoBARRIER, 2024-02-20:
What do you mean by manual calculix keyword edits?

FEA-eng, 2024-02-20:
[This button](https://wiki.freecad.org/FEM_SolverControl) allows you to write the input file and then edit it, modifying the CalculiX keywords used for analysis. The syntax is described in [CalculiX documentation](http://www.dhondt.de/ccx_2.21.pdf). When you're done with edits, save the file and run the analysis.

FEA-eng, 2024-02-27:
@maxwxyz Can you close this as not planned? The discussion can still be continued here or (better) on the forum but there's nothing to fix in FreeCAD.

closed 2024-02-27 by maxwxyz