FreeCAD/FreeCAD #12353: [Problem] Tree order is not respected

YosypC, 2024-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's like a feature in the tree is applied before the previous ones.
In the video I show that **Chamfer002** seems to apply apply right after **Chamfer**, since **Pad005** and **Chamfer001** are completely ignored. The tip was set at the latest feature for Chamfer002, it's the second time it happened in the session, the first time I had to delete the latest features and reapply them.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35315 (Git) AppImage
Build type: Release
Branch: main
Hash: 09f67f2f6bff1c57ac8019a5ba2f4a2c41025ee6
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

None

### Anything else?

[Appendiabiti (copy).zip](https://github.com/FreeCAD/FreeCAD/files/14239041/Appendiabiti.copy.zip) Renamed format

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-12:
@YosypC I'm not sure what's the problem, can you describe the steps to reproduce it?

YosypC, 2024-02-14:
I didn't narrow them down. I just applied features like chamfers and fillet.

In other words, in my case the latest chamfer (Chamfer002) in the tree seems to ignore both the pad and the previous chamfer above. If I select one of them as visible, the model is normally displayed with all the features in the history (obviously excluding the past ones, in this case Chamfer002), but as soon as I select the latest Chamfer002, it's like the tip of the tree was set at another step (in this case, internal name "Chamfer") and it doesn't display the two features above, while still being below them in the tree.

closed 2024-02-14 by maxwxyz

YosypC, 2024-02-15:
>  Maybe something in the process messed it up.

That's the point, it should not happen regardless. I'm sure I did nothing wrong, it also happened before and of course, deleting it and redoing it fixes the model. But it doesn't fix the problem of the tree not respecting the view, and vice versa.

There should be absolutely no instance in which a user is presented something different than the tree shows, even if "he does something wrong" (AKA not intuitive). Setting the tip elsewhere respects it because the additional steps are inserted right after the tip, before any other feature in the tree, following the hierarchy. But in this case, it doesn't, and it serves no purpose other than confusing the user.

YosypC, 2024-02-29:
@maxwxyz can this be reopened?

maxwxyz, 2024-02-29:
I'm not sure what's exactly the problem as I cannot reproduce. Maybe this is related to the TNP. If you have a different example just let me know how to reproduce it, the provided model was corrupt at a different feature in the tree. Just comment when you have the same problem the next time.

YosypC, 2024-02-29:
I forgot the model was broken, this one should be okay.
[Appendiabiti (copy).zip](https://github.com/FreeCAD/FreeCAD/files/14451791/Appendiabiti.copy.zip)
Opened with:
```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36220 (Git)
Build type: Unknown
Branch: main
Hash: 68fd2934cfcdb7b31d235d562412a5c04f55682e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenDark 2023.12.17
  * ThreadProfile 1.89.0
```

I haven't been able to find a way to reproduce it yet.