FreeCAD/FreeCAD #12361: [Problem] (help needed) Suppressed objects are not distinguishable

FlachyJoe, 2024-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

#12096 introduce a Suppressed property for PartDesign features and #12359 aims to add it for FEM constraints.
The tree items should reflect the suppressed status i.e. the tree item text should be striked through.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35932 +3 (Git)
Build type: Release
Branch: fem_suppress
Hash: 33634517a7a63a8b0b051a1e62f41de2a8ae2c51
Python 3.11.8, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-13:
Another way to indicate that a feature is suppressed is to add a red cross (or [No symbol](https://en.wikipedia.org/wiki/No_symbol)) on the left from it. Or gray it out. Strikethrough name makes sense too though.

maxwxyz, 2024-02-13:
the command is currently not present in the context menu or menu bar, right?

FlachyJoe, 2024-02-13:
> red cross

Couldn't too many small icons deserve the cause?

> gray it out.

It's already used for hidden objects.

> not present in the context menu or menu bar

Yes, I'm waiting the TNP solve to introduce a GUI command because of the risks of bad feedback. Nevertheless the (hiddden) property is already usable.

maxwxyz, 2024-02-13:
as a proposal, could the feature icon be crossed with a red line, similar as the hidden line over the eye icon here: https://github.com/FreeCAD/FreeCAD/pull/12298
additionally strikethrough text or display it in italic.

maxwxyz, 2024-02-13:
@FlachyJoe following up from the DWD discussion the favor is  grayed out + strikethrough text, to not introduce more clutter in the tree view.

FlachyJoe, 2024-02-13:
@maxwxyz Great, but I don't know how to change the tree display…

FlachyJoe, 2024-02-13:
I dug the code and found `ViewProviderAttachExtension::extensionMergeColorfullOverlayIcons` which do what we want, at least for icon.
A SuppressExtension seems a good idea as the Suppress feature has to be available throw many workbenches. Not sure I'm able to write this alone.

wwmayer, 2024-02-14:
```
from PySide import QtWidgets
tree = QtWidgets.QTreeWidget()
item = QtWidgets.QTreeWidgetItem()
item.setText(0, "Suppressed")
font = item.font(0)
font.setStrikeOut(True)
item.setFont(0, font)
tree.addTopLevelItem(item)
tree.show()
```

FlachyJoe, 2024-02-14:
@wwmayer Do you tell I have to use python to update the PartDesign objects at property change? :open_mouth:

wwmayer, 2024-02-14:
> Do you tell I have to use python to update the PartDesign objects at property change?

It's my reply to this question:

> do you know if strikethrough text in the treeview to indicate state of a feature is possible?

My answer gives a code snippet to demonstrate that it's possible. If it should become part of the core system then it will be done in C++, of course. I have written the code in Python because it's easier to confirm that it works.

FlachyJoe, 2024-02-14:
@wwmayer thank you.
I'm trying the Extension way :  #12412

PR #12412 (2024-02-26): Core: Create a SuppressibleExtension to provide the Suppressed property to all the WBs
Many objects can be suppressed, this extension aims to provide a unique visualization in the tree view.

Fix #12361
closed 2024-02-26 by chennes