FreeCAD/FreeCAD #12362: Techdraw performance issues

blinking-led, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

HI, everyone!

When I try to work with the Techdraw WB, it hangs with a high-load CPU. Techdraw works with two frames per minute. You can see it in the screenshot. When I delete Techdraw FreeCAD works well. Any suggestion? 
Intel core i5 1035G / 16 Gb / SSD

[image]
[image]
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-13:
@blinking-led can you try out the performance of the dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2024-02-13:
I have compiled _FreeCAD_ with _gprof_ and pressed some buttons in _PartDesign_, _Sketcher_ and more buttons in _TechDraw_. The output says that _FreeCAD_ spends time comparing `std::string`.

Perhaps, this is the right time to make use of `std::string_view`.

[output.txt](https://github.com/FreeCAD/FreeCAD/files/14268733/output.txt)

---

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35936 (Git)
Build type: Debug
Branch: main
Hash: b39d246c27ccbec0a503cd2d5e6dc08762951300
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

maxwxyz, 2024-02-13:
@WandererFan can you help out here?

WandererFan, 2024-02-13:
Whenever I've run TechDraw under valgrind, the majority of time is spent deep in the bowels of occ calculating intersections.  

Off hand, I can't think of anywhere TechDraw does enough string comparisons to compete with that.   Maybe the same comparison a ridiculous number of times? 

The picture looks like a circuit board.  Those are often slow to build due to the large number of faces to be checked.

You can use CoarseMode, set the Page to not update or turn off face finding to speed things up until you are ready to produce a finished drawing.

xtemp09, 2024-02-14:
Fine, I'll try Google's Orbit or tweak some options to get more useful information about the performance. But I'm right that the power of `std::string_view` (`QStringView`, `std::span` and other) must come to the abode of the source code of _FreeCAD_. At the moment, _FreeCAD_ uses it only in a handful of cases.

blinking-led, 2024-02-14:
Thank you for answering!
Except for night builds, I attempt to follow all advice. That was ineffective. I saw a similar freeze in today's test, which is simpler and involves about 20 objects in the drawing.
I'm using standard models from official websites, like https://www.dkceurope.com/en/prodotto/ramblock/cqe-modular-cabinets-with-blank-door-and-front-mounting-plate-2000x800x500-r5cqe2085t
file from the website [R5CQE2085.zip](https://github.com/FreeCAD/FreeCAD/files/14275986/R5CQE2085.zip)

[WandererFan](https://github.com/WandererFan) About "large number of faces" I made a "Simple copy" of DSK cabinets in the Part WB, and it changed the situation in the right way, but not enough. There might be a problem, while using TechDraw WB exclude calculation faces that are placed behind the front face?

blinking-led, 2024-02-15:
Today, I tried night build, but without any changes.
In the log panel, I see:

> 15:38:57  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: Access violation
> 15:56:30  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: Access violation
> 15:59:31  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: Access violation

blinking-led, 2024-02-19:
Today, I'm trying to nightly build:
```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
Wonderful! I make techdraw, but the UI shows only blank templates. After about 1–2 minutes, drawings appear.
Along with it, I see on the console:
> 09:06:14  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: Access violation
> 09:07:21  Unhandled std::exception caught in GUIApplication::notify.
> The error message is: invalid vector subscript
> 09:07:24  DrawProjectSplit::scrubEdges - OCC fuse raised warning(s):
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertSelfInterferingShape
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertBadPositioning
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertAcquiredSelfIntersection
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge
> BOPAlgo_AlertTooSmallEdge

WandererFan, 2024-02-19:
I can't be sure without a test file, but those BOPAlgo_Alert messages look like something you'd get if the input shape has problems.

@pavltom - might need your help here.

pavltom, 2024-02-19:
Hello, I have tried to import the **R5CQE2085.step** file. The model is pretty detailed and I am running a debug build of FreeCAD, nevertheless for me it loaded in about 30s. I have inserted few views from different angles into an A4 page, but although the rendering consumed about 2 minutes, it always produced an usable part view. I have scaled the view to fit the page, but again, so far no issues. The UI is not much responsive (sluggish at least), but there is really a lot of faces/edges to process.

@blinking-led maybe could You provide a saved document, which is causing You the troubles? Or some step-by-step scenario with the **R5CQE2085.step** file to reproduce the issue? I know this is rather a cliche, but on my machine everything works correctly ;-)

blinking-led, 2024-02-20:
Additional information. When I am trying to make holes in the body. The body has disappeared, and I see the next logs:

> 10:50:36  Unhandled exception in ViewProvider::eventCallback: Access violation
> (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
> 10:50:36  Unhandled exception in ViewProvider::eventCallback: Access violation
> (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
> 10:50:36  Unhandled exception in ViewProvider::eventCallback: Access violation
> (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
> 10:50:36  Unhandled exception in ViewProvider::eventCallback: Access violation
> (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
> 10:59:52  Hole: Linked shape object is empty
> 11:03:50  Hole: Linked shape object is empty
> 11:04:26  Unhandled Base::Exception caught in GUIApplication::notify.
> The error message is: Linked shape object is empty
> 11:05:11  pyException: Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> <class 'RuntimeError'>: Linked shape object is empty
> 11:05:29  Hole: Linked shape object is empty
> 11:05:31  pyException: Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> <class 'RuntimeError'>: Linked shape object is empty
> 11:06:08  Pocket: Linked shape object is empty
> 11:06:08  Pocket: Linked shape object is empty

blinking-led, 2024-02-20:
@pavltom Thank you! File attached.
[FOR_DEBUG.zip](https://github.com/FreeCAD/FreeCAD/files/14343659/FOR_DEBUG.zip)

Found one more bug in techdraw WB? Look at screenshoot
[image]

blinking-led, 2024-02-21:
"Scale to page" in TechDraw works uncorrect with the file 
> [R5CQE2085.zip](https://github.com/FreeCAD/FreeCAD/files/14275986/R5CQE2085.zip)
If I add some views and after that change in scale, the projection views become too far from each other.

blinking-led, 2024-02-26:
Periodically, I see a huge frame over the main window. It hides when I click on it.

[image]

maxwxyz, 2024-02-26:
@blinking-led Yes please close this if irrelevant and make another with the bug.

closed 2024-02-27 by blinking-led

frank-trampe, 2024-03-18:
@blinking-led, did you file a new bug? I am having a similar problem with periodic hanging accompanied by log output as follows.

```
18:50:18  DrawProjectSplit::scrubEdges - OCC fuse raised warning(s):
BOPAlgo_AlertBadPositioning
BOPAlgo_AlertSelfInterferingShape
BOPAlgo_AlertBadPositioning
BOPAlgo_AlertSelfInterferingShape
BOPAlgo_AlertBadPositioning
BOPAlgo_AlertSelfInterferingShape
BOPAlgo_AlertBadPositioning
BOPAlgo_AlertSelfInterferingShape
```

WandererFan, 2024-03-31:
The bad positioning and self interfering shape messages indicate that there are problems with the geometry of the source objects.

OCC really does not like problems with geometry.

pavltom, 2024-04-01:
I was thinking about dumping these OCC warnings on the Log level, instead of Warning level. I guess it can be quite confusing for a common user and as we can see above, it is unnecessarily flooding the log. What's your opinion?

frank-trampe, 2024-04-01:
Is there any way to determine which element is triggering those warnings?