FreeCAD/FreeCAD #12363: [Problem] Crash when editing sketches on master

andir, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've building FreeCAD from source for many months now. Mostly to help spot problems early. Perhaps I've finally managed to spot one such instance.

In my project there are several sketches linked into Part Design objects via Binders and I also make use of Spreadsheets for parameters (across documents). Whenever I entered one of my sketches (in the same document), did a minor edit (or sometimes just exiting the sketcher again) FreeCAD would crash. This wasn't happening with all the Sketches but some "older" (or further up in the chain).

The backtrace for the crash I observed is quoted below. I've had a skim through the Git log between the versions that I tried and reverting 4d929d33017728549f17e527e6ec7507ffb166e3 & 86e75037d9763466e7f9cc6cbf7ca4fff20eabac.  made the problem not reproducible (for now). They are also the only changes touching related code (that I could find) since my last build.

I don't fully understand the code in 4d929d33017728549f17e527e6ec7507ffb166e3 but perhaps the baseShape could be a nullptr and that not being expected?

```
gdb-peda$ bt
#0  Base::BaseClass::isDerivedFrom (this=0x0, type=...) at /build/source/src/Base/BaseClass.h:145
#1  0x00007f7da031ac82 in PartDesign::Feature::getBaseShape (this=<optimized out>) at /build/source/src/Mod/PartDesign/App/Feature.cpp:159
#2  0x00007f7da031ba8e in PartDesign::Feature::recompute (this=0xcb483e0) at /build/source/src/Mod/PartDesign/App/Feature.cpp:68
#3  0x00007f7f6afdf82d in App::Document::_recomputeFeature (this=this@entry=0xa13e770, Feat=0xcb483e0) at /build/source/src/App/Document.cpp:3129
#4  0x00007f7f6afe5e17 in App::Document::recompute (this=this@entry=0xa13e770, objs=std::vector of length 0, capacity 0, force=<optimized out>, hasError=hasError@entry=0x0, options=options@entry=0x0) at /build/source/src/App/Document.cpp:2856
#5  0x00007f7f6b0b2668 in App::DocumentPy::recompute (this=this@entry=0xa0d3e30, args=<optimized out>) at /build/source/src/App/DocumentPyImp.cpp:584
#6  0x00007f7f6b0b293a in App::DocumentPy::staticCallback_recompute (self=0xa0d3e38, args=<optimized out>) at /build/source/build/src/App/DocumentPy.cpp:1952
#7  0x00007f7f69f37ce2 in cfunction_call () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#8  0x00007f7f69eee3ed in _PyObject_MakeTpCall () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#9  0x00007f7f69de38dd in _PyEval_EvalFrameDefault () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#10 0x00007f7f6a01856c in _PyEval_Vector.constprop.0 () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#11 0x00007f7f6a01870a in PyEval_EvalCode () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#12 0x00007f7f6a043430 in run_mod () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#13 0x00007f7f6a043ac3 in PyRun_StringFlags () from /nix/store/y027d3bvlaizbri04c1bzh28hqd6lj01-python3-3.11.7/lib/libpython3.11.so.1.0
#14 0x00007f7f6a91b465 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) (this=<optimized out>, sCmd=sCmd@entry=0x1e47d418 "App.ActiveDocument.recompute()") at /build/source/src/Base/Interpreter.cpp:249
#15 0x00007f7f6bd0230d in Gui::Command::_runCommand (file=file@entry=0x7f7d15a921c0 "/build/source/src/Mod/Spreadsheet/Gui/SheetModel.cpp", line=line@entry=0x229, eType=eType@entry=Gui::Command::Doc, sCmd=0x1e47d418 "App.ActiveDocument.recompute()")
    at /build/source/src/Gui/Command.cpp:725
#16 0x00007f7f6bd024ac in Gui::Command::_doCommand (file=file@entry=0x7f7d15a921c0 "/build/source/src/Mod/Spreadsheet/Gui/SheetModel.cpp", line=line@entry=0x229, eType=eType@entry=Gui::Command::Doc, 
    sCmd=sCmd@entry=0x7f7d15a8d978 "App.ActiveDocument.recompute()") at /build/source/src/Gui/Command.cpp:677
#17 0x00007f7d15a078c6 in SpreadsheetGui::SheetModel::setCellData (this=this@entry=0xb725b80, index=..., str=...) at /build/source/src/Mod/Spreadsheet/Gui/SheetModel.cpp:553
#18 0x00007f7d15a07ced in SpreadsheetGui::SheetModel::qt_static_metacall (_o=0xb725b80, _c=<optimized out>, _id=<optimized out>, _a=0x1df1c1c8) at /build/source/build/src/Mod/Spreadsheet/Gui/SpreadsheetGui_autogen/include/moc_SheetModel.cpp:76
#19 0x00007f7f680ebf20 in QObject::event(QEvent*) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#20 0x00007f7f68b97dce in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Widgets.so.5
#21 0x00007f7f6bcd23c1 in Gui::GUIApplication::notify (this=0x7ffff6b70180, receiver=0xb725b80, event=0x1df1c180) at /build/source/src/Gui/GuiApplication.cpp:83
#22 0x00007f7f680bfc58 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#23 0x00007f7f680c32f1 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#24 0x00007f7f681199d3 in postEventSourceDispatch(_GSource*, int (*)(void*), void*) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#25 0x00007f7f63b23744 in g_main_dispatch () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#26 0x00007f7f63b26797 in g_main_context_iterate_unlocked.isra () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#27 0x00007f7f63b26d8c in g_main_context_iteration () from /nix/store/hbdl12ck2ibhn5zsi8nsfa6v471pf0j6-glib-2.78.4/lib/libglib-2.0.so.0
#28 0x00007f7f681191b6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#29 0x00007f7f680be67b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#30 0x00007f7f680c6ef0 in QCoreApplication::exec() () from /nix/store/znk677nrcipl9jrnwjfzfx65wdmh77iy-qtbase-5.15.12/lib/libQt5Core.so.5
#31 0x00007f7f6bc2caad in Gui::Application::runApplication () at /build/source/src/Gui/Application.cpp:2425
#32 0x0000000000415d26 in main (argc=<optimized out>, argv=<optimized out>) at /build/source/src/Main/MainGui.cpp:275
#33 0x00007f7f6783a0ce in __libc_start_call_main () from /nix/store/cyrrf49i2hm1w7vn2j945ic3rrzgxbqs-glibc-2.38-44/lib/libc.so.6
#34 0x00007f7f6783a189 in __libc_start_main_impl () from /nix/store/cyrrf49i2hm1w7vn2j945ic3rrzgxbqs-glibc-2.38-44/lib/libc.so.6
#35 0x0000000000415605 in _start ()
```

### Full version info

```shell
Git revision: a1fcb665368edf04609747079b4a9cd30c4f21ec

Build information from FreeCAD:

OS: NixOS 24.05 (Uakari) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown
Build type: Release
Python 3.11.7, Qt 5.15.12, Coin 4.0.1, Vtk 9.2.6, OCC 7.8.0
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.72
  * Assembly4 0.50.6
  * Curves 0.6.15
  * A2plus 0.4.61
  * CurvedShapes 1.0.5
  * ProDarkThemePreferencePack 1.0.0
  * ExplodedAssembly
  * parts_library
```
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-02-13:
Hi,
could you please share the file?

andir, 2024-02-13:
Here you go: https://s.rammhold.de/freecad-crash-example.zip

The file is unfortunately too large for GitHub to accept it.

Step to reproduce: Open the "fuse-example.FCStd" file and open the Sketch003, potentially try editing the radius of one of the rounded rects. It is often enough to crash it for me.

PR #12385 (2024-02-13): PD: handle null pointer in Feature::getBaseShape
Fix #12363
closed 2024-02-13 by adrianinsaval