FreeCAD/FreeCAD #12365: [Problem] UCNC Post Processor drilling error

jabenetti, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

G-Code G81 drilling canned cycle only drills the first hole in a multi hole operation

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Argentina (es_AR)
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Schildkroet, 2024-02-14:
Which post processor are you using? uccnc?
Can you please provide an example file + output?

jabenetti, 2024-02-15:
I'm using uccnc as post processor. The problem is than depth is only used in first hole with G81 code:
G81 Z-10.000 R4.000 F100.000    ( FIRST HOLE, OK )
G0 X18.710 Y0.000
G81 F100.000                               ( SECOND HOLE, WRONG, NO Z DEPTH )
G0 X0.000 Y19.678
G81 F100.000                               ( THIRD HOLE, WRONG, NO Z DEPTH )
G0 X-19.812 Y0.000
G81 F100.000                               ( FOURTH HOLE, WRONG, NO Z DEPTH )

Schildkroet, 2024-02-15:
If you add "--repeat" to args, it works.

It seems REPEAT_ARGUMENTS filters out Z-Pos because it doesn't change for the drilling cycles. There should be added an exception for canned drilling cycles.

Schildkroet, 2024-02-15:
I created a PR. Please test it for yourself @jabenetti

PR #12426 (2024-02-19): [Path] Bugfix uccnc_post.py
Fixed canned drill cycles for uccnc post. Fixes #12365
closed 2024-02-19 by sliptonic