FreeCAD/FreeCAD #12369: [Problem] TechDraw shouldn't need a 'refresh page' button.

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDraw shouldn't need a 'refresh page' button. Why not using the existing core refresh button instead?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-02-15:
Refresh should be automatic, perhaps with a global toggle to disable automatic refresh as needed within complex documents.

PaddleStroke, 2024-02-19:
Thanks for your feedback. I opened these issue as I discovered the workbench so I noted everything that seemed not ideal, but maybe they don't all make sense. Indeed if it refresh only the page then it's different than the recompute.
Though the page seemed to refresh automatically when I was testing isn't it the case?

closed 2024-02-20 by PaddleStroke