FreeCAD/FreeCAD #12371: [Problem] TechDraw annotation shouldn't need a dialog

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TechDraw annotation dialog UI is very strange. Why not simply using line breaks instead of this exotic + - buttons?

Besides why a dialog in the first place? The user should be able to start edition directly on the annotation.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-02-19:
The 2 tools are essentially "quick, casual, plain text annotation" and "full featured formatted text annotation".

WandererFan, 2024-02-19:
This commented code in QGIViewAnnotation should implement "edit directly on the annotation",  but it is 8 years old, so no guarantees.
```
    m_textItem->setTextInteractionFlags(Qt::NoTextInteraction);
    //To allow on screen editing of text:
    //m_textItem->setTextInteractionFlags(Qt::TextEditorInteraction);   //this works
    //QObject::connect(QGraphicsTextItem::document(), SIGNAL(contentsChanged()), m_textItem, SLOT(updateText()));  //not tested
```