FreeCAD/FreeCAD #12373: [Problem] TechDraw print all pages should be in file menu

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Print All pages would be expected to be located in the File/Print all pages menu. It is bloating in the toolbar.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-20:
@WandererFan thoughts ?

PaddleStroke, 2024-03-20:
These serie of ticket was my initial reaction to the numerous icons in techdraw. I'm not sure yet how valid they are.

@obelisk79 perhaps this could be a discussion matter for the DWG?

obelisk79, 2024-03-20:
So, when loading up in 1080p and default icon size, I'm not sure the toolbars are a significant issue here. However there is most likely room for some optimization. I can bring this up with the DWG, but I'm juggling a few projects right now and my time to spend on an in-depth review is very limited right now.

Here is how it currently looks. I don't see any reason why we couldn't move this to the File menu or print dialog since it is universal across desktop applications to access printing from there. However the toolbutton seem logical also, this comes down to a matter of taste I think.

[image]

luzpaz, 2024-12-02:
tangetially related:  
* https://github.com/FreeCAD/FreeCAD/issues/18235