FreeCAD/FreeCAD #12378: [Problem] TechDraw : Stack group could be removed if tree reordering is enabled.

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Stag Group tool to move rendering of views up / down could be removed after tree reordering merges. Then we can just render based on the tree order and let user reorder there easily.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-13:
https://github.com/FreeCAD/FreeCAD/pull/12293

maxwxyz, 2024-03-02:
@PaddleStroke I've tested it but it did not work by drag and drop in TechDraw to reorder items.
I have also trouble with reordering items from Part WB and will create another issue with a demo.

maxwxyz, 2024-05-25:
@PaddleStroke I think I can close this, right?

PaddleStroke, 2024-05-25:
Closing it is ok for me.
Though I would think that these actions would be better as right click menu
rather than in the toolbar.
It would feel more natural to me.

closed 2024-05-25 by maxwxyz