FreeCAD/FreeCAD #12379: [Problem] TechDraw : "Add view to clip group" should not be a tool.

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

"Add/Remove view to clip group" should not be a tool. It should be done by drag and drop.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-03-23:
The only time I ever use clip group is when there is a bug report for clip group.   I'm not sure what it does that couldn't be done with detail view (which didn't exist when clip group was ported from DrawingWB).

Should get some user f/b on this.  I'll post a link in the forum.

WandererFan, 2024-03-23:
forum link:  https://forum.freecad.org/viewtopic.php?t=86362

PR #13063 (2024-03-25): TechDraw: Fix drag&drop of views. Add auto-switch to techdraw on double click.
This PR address 3 low hanging fruits in techdraw : 
- Fixes #13061
In sketcher double clicking on a sketch changes the active workbench to sketcher.
Similarly double clicking on an assembly switch to assembly.
For consistency, I offer to implement the same behavior when double clicking on a page, switch to techdraw.

- Fixes #12375 :
Techdraw has a 'move view' tool that is pointless. Moving views should be done by drag and drop. It actually was supported before but got broken with the introduction of `canDropObjectE [... cut]
closed 2024-03-25 by WandererFan