FreeCAD/FreeCAD #12382: Updates cause 'scavenger hunts' for features/controls [Problem]

rv7charlie, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changes to the locations of functions/features/controls make it difficult to impossible for 'newbies' to follow existing tutorials. 

### Full version info

```shell
If significant changes to locations are implemented, there should also be some obvious and easy-to-access method of finding those functions/features/controls. My immediate example is the removal of the 'edit controls' subpanel from the Combo>tasks panel. The release notes for 0.21 simply says, " This PR also removes the "Edit controls" widget from the task panel as all it's contents have been moved to other places or removed. " Trying to follow a tutorial that references these controls has become difficult, at best. (This isn't the only example; just the immediate one for me.) 
Suggestion, for this problem *and* for usability in general:
A 'Find' function, built into FreeCAD itself, would be a godsend. Most software packages offer a 'find' option. It took me many minutes to figure out what the arrow/question mark button did, since "what's this?" seemed to stimulate a denial response when I pointed at any feature of a drawing. A 'find feature' button would be great when, for instance, I wanted to use the 'mirror' feature I've read about but don't know the hieroglyphic for, or even which toolbar it's on.

Sorry for the rather 'wordy' explanation, but trying to give some idea of what newbies to CAD are experiencing.

Having said all that, please don't mistake my complaint for ingratitude. I'm loving the fact that I finally have the chance to learn CAD at age 72. FreeCAD really is an impressive body of work, and I'm grateful it exists!
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-13:
Documentation is a continuous effort and is worked on. The suggested feature to _find_ commands was already discussed somewhere.

maxwxyz, 2024-02-13:
@rv7charlie there is another issue similar to your request so I'm closing this issue as duplicate of #8290 
You can comment there as well.

closed 2024-02-13 by maxwxyz

rv7charlie, 2024-02-13:
Good to know it's being considered. Thanks for your quick response, and
thanks for what I know is a massive project, keeping FreeCAD up and running.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>