FreeCAD/FreeCAD #12383: [Problem] TechDraw: Dimensions are those of projected geos.

PaddleStroke, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Am I missing something? Dimensions tools write the dimension of the projected geos. So if the view is not normal it is wrong.

[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-13:
That's the overall behavior of TD. For axonometric views there is the axonometric dimension tool. You can also link the dimensions to 3D objects or overwrite them manually.

obelisk79, 2024-02-15:
I agree with paddle. This is a reasonable address where some modest automation can be applied.

WandererFan, 2024-02-19:
You can't get a 3d ref from a 2d ref.  Projection is not an invertible operation.   

At best, you could scan every 3d vertex (or edge as the case may be), project it, then compare each to the 2d ref.   In the general case, this will give a list of potential matches. You would still need manual intervention to pick the right match. 

There is nothing stopping the user from creating dimensions directly from 3d model geometry.   This will create a True dimension, rather than a Projected dimension

WandererFan, 2024-02-19:
This issue should be closed.

maxwxyz, 2024-02-19:
I will close this then

PaddleStroke, 2024-02-20:
I disagree that this should be closed. The user can litteraly shoot himself in the foot by creating non-sensical dimensions, no warning. It is very misleading.
I don't think other CAD softwares enable you to create such non-sense dimensions. I don't know for all but in catia for example : 
[image]
I would be surprise solidworks let you do it either.
Please re-open the issue

maxwxyz, 2024-02-20:
@WandererFan is it possible to distinguish from the view if the view is axonometric or not to automatically choose the right dimension?

WandererFan, 2024-02-24:
You can use the the view direction to try to determine if the view seems to be axonometric (ex (1,1,1)) but the orientation of the subject object may mean that what looks like an axonometric view according to the direction is actually orthogonal due to the alignment of the subject.

If you insist on choosing your dimension points on the drawing instead of selecting 3d geometry you are going to get projected dimensions. 

There is no way to convert a projected point back to its unprojected source.

WandererFan, 2024-02-24:
This is dead easy in Sketcher since sketcher views are 2 dimensional and nothing is projected.

PaddleStroke, 2024-02-24:
There is no way to convert projected back to original, but it is possible
to store a map of projected-original edge.

PaddleStroke, 2024-02-24:
With a map of projected-original edges, it should be no problem.

howie-j, 2024-02-24:
Onshape for reference:
[image]

pieterhijma, 2024-03-05:
> You can't get a 3d ref from a 2d ref. Projection is not an invertible operation.
>
> At best, you could scan every 3d vertex (or edge as the case may be), project it, then compare each to the 2d ref. [...]

> There is no way to convert projected back to original, but it is possible to store a map of projected-original edge.

I would like to react on these two comments from @WandererFan (the first) and @PaddleStroke (the second). I tried to modify OCCT to maintain a mapping between the offered 3D model and the returned 2D projection. I don't think it can be solved in general, but I think there is potential to do much better than what is possible right now. See [this forum post](https://forum.freecad.org/viewtopic.php?p=697345#p697345).

WandererFan, 2024-03-13:
IDK, but I'm told that dimensions starting as the projected value is the way other CAD programs work.  @howie-j  picture seems to bear this out.

maxwxyz, 2024-03-13:
I think how other programs handle this is that you define the dimensions in the 3D view (e.g. annotations) before deriving the projected drawing. Then the dimensions get projected also on the sheet and could be linked. If you then change them, they are also changed in the 3D view. I know that's possible in SolidEdge and here are some Inventor references [1-3].

This would in a far future be great for a real GD&T implementation, as they are defined also in the 3D space as annotations on the corresponding features of the model and can then be derived into the 2D drawing.

[1] https://help.autodesk.com/view/INVNTOR/2022/ENU/?guid=GUID-597185E4-F39C-4CE7-9CC7-EA69533BACB2
[2] https://www.cadforum.cz/en/controlling-size-of-3d-model-by-changing-drawing-dimensions-tip10521

WandererFan, 2024-11-02:
just as documentation...

discussion on dimensions in axo views: https://www.eng-tips.com/threads/is-dimensioning-detailing-in-isometric-view-acceptable.157995/

a google search for "dimension axonometric view" is also informative wrt how other CADs handle axo dims.

getting the 3d measurement onto the page seems to require an additional step in most cases.

closed 2024-11-02 by WandererFan

FEA-eng, 2024-11-04:
May I ask why this was closed as completed?