FreeCAD/FreeCAD #12384: [Problem] [Sketcher] When floating widgets are enabled, the auto constraint doesn't select the diameter constraint, when creating a circle

Be-Cool-Man, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Step 1:** Enable »On-View-Parameters« in [Edit] ⇒ [Preferences …] ⇒ [Sketcher] ⇒ [General] ⇒ [Tool Parameters]
**Step 2:** Open the Sketcher Workbench, and create a new sketch (The exact working plane doesn't matter)
**Step 3:** Create a circle (The exact location doesn't matter)

**Observed behavior:** The auto constraint uses the radii constraint for circles and for arcs. And therefore, it ignores the settings in [Edit] ⇒ [Preferences …] ⇒ [Sketcher] ⇒ [General] ⇒ [Dimension constraint] ⇒ »Dimension tool diameter/radius mode: Auto«

**Expected behavior:** The auto constraint respects the settings in [Edit] ⇒ [Preferences …] ⇒ [Sketcher] ⇒ [General] ⇒ [Dimension constraint] ⇒ »Dimension tool diameter/radius mode: Auto« and therefore uses the diameter constraint for circles and radii constraints for arcs.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35927 (Git) AppImage
Build type: Release
Branch: main
Hash: 9c3ab15ecbe4d904330ff5202f488d33e0dd841f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * Assembly4 0.50.6
  * A2plus 0.4.62
  * BOLTSFC
  * ExplodedAssembly
  * Assembly3 0.12.2
  * sheetmetal
  * dodo 1.0.0
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Curves 0.6.22
  * OpenDark 2023.12.17
  * fasteners
  * Help 1.0.3
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

koyashi, 2024-02-16:
Can confirm the same behaviour is happening with the Windows version.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35927 (Git)
Build type: Release
Branch: main
Hash: 9c3ab15ecbe4d904330ff5202f488d33e0dd841f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

maxwxyz, 2024-02-17:
@PaddleStroke

maxwxyz, 2024-03-01:
I'll add from #12681 it should be possible to manually switch the dimension type (radius or diameter) for arcs in the OVP when drawing them.

pierreporte, 2024-03-01:
@maxwxyz it should be possible for circles as well.

commit a4b97e29 (2024-06-03): Fixes #12384
closed 2024-06-03 by chennes