FreeCAD/FreeCAD #12387: [Problem] Techdraw > plot > source mirrored objects not show

bjnobre, 2024-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Techdraw do not show the source element of a "part workbench mirror feature":
[image]
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.62
  * sheetmetal 0.2.63
[/code]

and

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35927 (Git)
Build type: Release
Branch: main
Hash: 9c3ab15ecbe4d904330ff5202f488d33e0dd841f
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.62
  * sheetmetal 0.2.63
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-02-13:
Need the file to tell you anything useful.    If I was to guess, I'd say that the Source property in the view is not what you think it is.

bjnobre, 2024-02-13:
I’ve just realize that (something with the source) and I’ve tested it on a new file and saw that I have to select as source all the bodies, not just the last one!

closed 2024-02-13 by bjnobre