FreeCAD/FreeCAD #12394: [Problem] Sketcher text rendering is "blurry" due to incorrect compositing creating dark fringe around letters

Nathan-Fenner, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The text used for sketcher dimensions looks "blurry" when the background is light and the text color is dark. Here's an example screenshot:

[image]

In the above, the background color is configured to be pure white (255, 255, 255) and the text color is pure red (255, 0, 0).

For comparison, here's text in a similar font/size and the same red color, added in an image-editing program:

[image]

- (A) is the text rendered by FreeCad
- (B) is the text I added in an image editor

Here's a 10x enlargement to show what's going on more clearly:

[image]

Note that the text rendered by FreeCAD is *darker* on the edges, while the text added in the image editor is just *light* because the red is blending into white.

It looks to me that FreeCAD is rendering the text with pre-multiplied alpha, and then compositing it onto the screen as if it is non-premultiplied alpha. I was able to achieve this effect manually in the image editor:

[image]

Ideally, the text would be composited correctly, so that it doesn't produce this dark fringe. This would make dimension constraint text a lot clearer and easier to read.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-06:
There was a fix for blurry fonts with https://github.com/FreeCAD/FreeCAD/commit/41e1626bd1561b0e629bce09f1aafe8df27bf23c but I don't see that particular line of code in the current master tree. Can anyone confirm this hasn't been lost?

I'm seeing horrible font rendering in Sketcher as well, but I believe that's because my default fonts are bitmaps. It would be nice to have a preference for font to use. Also, it seems that the text is rendered in 3D, which I find to be problematic. Why would we want constraint text rendered on a plane that isn't our display?

Nathan-Fenner, 2024-05-06:
> There was a fix for blurry fonts with https://github.com/FreeCAD/FreeCAD/commit/41e1626bd1561b0e629bce09f1aafe8df27bf23c but I don't see that particular line of code in the current master tree. Can anyone confirm this hasn't been lost?

The shape of the text is actually correct (see the 10x enlargement above for comparison - each pixel is located exactly where it's supposed to be, there's no misalignment). So the linked fix would not affect this issue one way or the other.

The problem is only in the color blending / compositing, which is causing the fringes of the letters to be darker than they're supposed to be (at least when composited onto a light background).

kriswilk, 2024-10-10:
I am shocked that this issue is not more prominent. The default combination of background and font rendering is a mess. Just look at this screenshot from vanilla 1.0RC2... the vertical dimension in particular is practically unreadable!

[image]

In my preferred theme (OpenTheme Light) it's better, but you can still see that the font is rendered with the odd "halo" that @Nathan-Fenner originally reported:

[image]

This is a very visible problem that affects every user and leaves a poor first impression. I hope that this can be investigated before the world takes a critical look at 1.0.

SkyAlexV, 2024-11-17:
Have similar issue, font in sketcher on dark background fully unreadable.
[image]

Nathan-Fenner, 2025-02-25:
I verified that there is no change in weekly build 40317, this is still an issue.

otede, 2025-05-16:
Still an issue and still the worst possible impression. Right in the first 30 minutes with FreeCAD user is thrown into a squinting game!

PR #21536 (2025-06-23): [GUI] Remove dark fringe around letters

Closes #12394. I have read too much of literature, tried many things, but so far, I cannot say why exactly this works.

`glBlendFunc(GL_ONE, GL_ONE_MINUS_SRC_ALPHA);` is the right way to blend colors with alpha channel, however, it gives blur with single-colored background, so conversion just ARGB32 without pre-multiplication gives good-looking result.

Just ARGB32 without pre-multiplication with unchanged blending parameters gives dark fringe anyway.

It would be good if somebody tests it on _Windows_ with Line Smoothing/Antialiasing on.
commit 65f737e8 (2025-06-23): [GUI] Remove dark fringe around letters (#21536)

Closes #12394
closed 2025-06-23 by chennes