FreeCAD/FreeCAD #12395: [Problem] Assembly: Dragging needs to be improved

PaddleStroke, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In assembly we have one generic move mode "Default mode". This mode simply move the mouse on the focal plane at the mouse position. While it works well with unconnected parts, as soon as the part is connected, it does not work smoothly and can break easily (crazy movements and positions of parts breaking).
To mitigate this issue, special move modes were introduced : 
- Rotate : Triggered when the part is connected to ground by a rotation joint
- Slider : Triggered when the part is connected to ground by a slider joint

We now need mode of such special movement for the following modes : 
- [ ] Cylindrical
- [ ] Ball
- [ ] For the various Distance joints

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-15:
What I want to add here: When selecting and dragging from the 3D view, nothing happens until you select the entire body in the 3D view or the body in the tree.

maxwxyz, 2024-02-16:
@PaddleStroke
another bug in dragging. When the view is perpendicular, the objects disappear completely and cannot be restored. Shall I create a separate issue for this?
Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/5abd8e11-ab84-47a1-b568-80904959377c

PaddleStroke, 2024-02-16:
I see, I guess the same behavior happens for revolute joint when view is perpendicular too. We need to put a failsafe there. Yes better make a separate issue

closed 2024-12-16 by PaddleStroke