FreeCAD/FreeCAD #12396: Transparent faces of an opaque part render properly on screen but export to .step renders as opaque.

parastream, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am designing a clear Plexiglas (Acrylic) panel. It's very simple, a rectangle with 8 small holes, pad to 0.25 inches. I want it to look mostly opaque around the edges. I found the best way to do this was to create the pad with no transparency and the set the top and bottom faces with the alpha channel set to 0. It looks perfect on screen, but when I export it to .step and look at it, it's solid.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-14:
so you mean the colors of the step are not exported as they are displayed in FC? Could you try this in the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-03-02:
@davesrocketshop will this be different with material overhaul or has this nothing to do with face color and alpha values?

davesrocketshop, 2024-03-02:
> @davesrocketshop will this be different with material overhaul or has this nothing to do with face color and alpha values?

This is not directly related to the material overhaul but will be affected by it. It would require fixes either way. It really is a problem with the export.

That said, I'd wait until I merge the updated appearance code, which should be within a couple of weeks. Otherwise it would have to be fixed again.

maxwxyz, 2024-06-28:
the appearance code is merged now, right?

maxwxyz, 2024-07-01:
@davesrocketshop @parastream is this still an issue now?

parastream, 2024-07-01:
Sorry about not getting back… No, I managed to create the face colors in FreeCAD so that they export the way I want in .step.

 

Robert

closed 2024-07-01 by maxwxyz