FreeCAD/FreeCAD #12397: [Problem] Assembly : Coordinate systems objects are bundle in joints

PaddleStroke, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The joints objects are using 'Joint Coordinate Systems' that are computing their location based on user selection.
Each joint has 2 JCS that are bundled in the joint object.

It would be better if the JCS object was a separate object, so that user can manipulate them and reuse them in other workbenches (in MBD for example)

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-02-14:
Would the coordinate system object be a child of the joint? Then it would be easier to find.

PaddleStroke, 2024-02-14:
Yes it would.
I'm still mitigated about doing this actually. Because it would bloat the tree and make things more complicated and I'm not sure of the benefit of being able to manipulate the JCS. The only usecase I have in mind is for MBD where they want to position some markers for the analysis. Thoughts?

pierreporte, 2024-02-14:
If you don’t want to bloat the tree, perhaps you could make the JCS inherit coordinates systems properties so that MBD could pick them directly. I don’t think that FreeCAD should accommodate for external workbenches but this one is bringing very desirable features that would be legitimate built-in. If there is a possibility to merge it (after all it’s in the scope of the assembly workbench), then this can be a good idea.

PaddleStroke, 2024-12-16:
I think with the core datum this one can be closed

closed 2024-12-16 by PaddleStroke