FreeCAD/FreeCAD #12398: [Problem] Assembly : No basic motions such as 'rack and pinion'

PaddleStroke, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great to have basic motions in assembly such as rack and pinion.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-14:
That would be nice. SolidWorks has the following types of mates:
- cam-follower
- gear
- hinge
- rack and pinion
- screw
- slot
- universal joint

I guess that gear mates would be particularly useful.

pierreporte, 2024-02-14:
Belts would be useful too. Basically rack and pinion but with inverted rotation.

maxwxyz, 2024-02-16:
In general a motion analysis is missing. Define end stops for motion types / joints, include collision as a stop, define certain angles how something can turn or bend. This allows to evaluate mechanisms and detect potential interferences during motion.

Should I create a new feature for this or is this enough here?

luisDias89, 2024-02-21:
These are some features that are missing in this workbench:
Example of Gear: https://www.youtube.com/watch?v=xey6WL873-E
                            https://www.youtube.com/watch?v=ejCoK0uaVYQ

Example of rack : https://www.youtube.com/watch?v=k-4FIJmwCRI:

luisDias89, 2024-02-21:
When you select two faces, two disks, edges, or other object items, it will analyze all the possibilities that could be achieved with them and will automatically provide a menu. Initially, the menu will be semi-transparent. If the mouse cursor is moved over the menu, it becomes available; if the mouse moves away, the menu disappears.
Some Features for example for planing is make coincident, make paralel, distance betwen a range, distance only, lock, perpedicular, range of angle betwen boot, or fixed angle.

Look at this image as an example:
image

Other example is two circles:
image
We have concentric, tangent, lock...

And so on, another example: among gears, you can define when one gear rotates in relation to another, making animations super realistic, with the teeth rotating perfectly meshed.

And many other options that can be integrated.

This is the most importante workbench for large assembly, this is what the freecad need to bee a great Software.

A table with relations betwen geometric properties: easy ->
image

All of them
image

luisDias89, 2024-02-21:
@PaddleStroke 
Exemple of Gear Assembly
Lock at the mouse, we can able to move object with mouse and see in real time one animation (min 5:39)
https://www.youtube.com/watch?v=JcOO6jcdZ_I

Watch this video from minute 24 - This is a Great Example how easy is one assebmly in other software. Freecad need This.
https://www.youtube.com/watch?v=VIC5ky8Pq8s&list=PLm1W92pCWeKNo5PqIIi7fn51-tWvzpPOv&index=8

PaddleStroke, 2024-03-16:
@qewer33 interested to work on motion icons? :D 
I think I will start with Gear, Rack and pinion and screw.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre