FreeCAD/FreeCAD #12399: [Problem] Regression on FreeCAD infos

PLChris, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The FreeCAD infos show  in the forum 3 quotes at the start and end:
[image]
Before they used to include the code tags:
[image]
Before that they showed just plain text.

Please return to the code tags or the plain text.

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35594 (Git)
Build type: Release
Branch: main
Hash: 49e7bba9c7390b11e6cc9acebd10cc1568ec8869
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.4.75
  * sheetmetal 0.3.14
  * ExplodedAssembly
  * Curves 0.6.21
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-02-14:
They are formatted for GitHub now: https://github.com/FreeCAD/FreeCAD/pull/11226

But if the forum format doesn't change, maybe plain text would be better. More people post on the forum than here, after all.

maxwxyz, 2024-02-14:
I bet there is already some discussion on the forums. If possible, the forum should be capable to handle the format as is.

Roy-043, 2024-02-14:
The current situation is akward. Pasting the triple quoted text causes a formating problem when you create a GitHub issue, as can be seen here. The current template adds additonal triple quotes.

yorikvanhavre, 2024-02-16:
I'd be in favor of pasting text only, with no formatting

closed 2024-03-03 by wwmayer