FreeCAD/FreeCAD #12400: [Feature] Make editable strings in the templates and text annotations configurable by spreadsheet and functions

chhoehn89, 2024-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The editable strings in the templates and the text annotations are static input strings. 

Please make them configurable by spreadsheet and by functions.

This is useful for example for configuration tables, where many drawings with individual texts can be derived from one part.

[image]
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60n
  * Curves 0.6.12
  * Defeaturing 1.2.0
  * freecad.gears 1.0.0
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-02-14:
@chhoehn89 any reason you aren't using latest stable 21.2.0 ?

chhoehn89, 2024-02-14:
Nope no specific reason.

pierreporte, 2024-02-14:
Related to #12135 and #12136.

yorikvanhavre, 2024-02-16:
A nice and easy way could be to dynamically create/manage object properties linked to the template editable fields...