FreeCAD/FreeCAD #12416: Enhancement: Optimize included png and svg files

rmeissn, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When heading through my local filesystem with image optimization tools (for svg, png and jpg files), I noticed that some freecad related files were optimized.

Thus I've downloaded the current source code and executed those programs within the folder.
Results:
- SVG files are compressible by about 40-90% with `svgo -r -f ./FreeCAD-main --multipass`
- PNG files are compressible by about 3-30% with `oxipng -o max -r -p ./FreeCAD-main`

Possibly apply those programs and optimizations to your image files to save some space, reduce download sizes and necessary computing power to display the images - even though all of these are just minimal improvements.

### Full version info

```shell
None
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-15:
@obelisk79 what do you think about this?

xtemp09, 2024-02-15:
@rmeissn, how much space have you saved?

rmeissn, 2024-02-15:
For the current source-code (downloaded as a zip and extracted, according to `du`):
Current size: 373 200 KB
Optimized Size: 356 704 KB

But it might be best to optimize these files after the build step, as e.g. `svgo` is removing data that is unnecessary for display, but might be valuable for further development.

luzpaz, 2024-03-20:
Any decisions on this ?

obelisk79, 2024-03-20:
I didn't think anything ever came out of this. I commend the efforts to optimize the images.