FreeCAD/FreeCAD #12418: [Problem] TechDraw: associating dimensions with 3D references moves the page dimensions to another TD's view's edge

YosypC, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/50027752/419b110e-ad6f-473c-8dda-bbc685edb925

https://github.com/FreeCAD/FreeCAD/assets/50027752/ffc36ce6-c9d9-45c3-bc17-dcbfb0cfd231

In case the messy title wasn't clear, when I try linking a View's edge on a TechDraw page with a model's edge in the 3D view, the edge in the TD view changes. I see no way of reassigning it to the initial edge.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git) AppImage
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.7
  * OpenDark 2023.12.17
  * A2plus 0.4.62 (Disabled)
```

### Subproject(s) affected?

Techdraw

### Anything else?

[Desk.zip](https://github.com/FreeCAD/FreeCAD/files/14298123/Desk.zip)
Raname in FC format.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YosypC, 2024-02-16:
Ondsel-Development#24 could be related

WandererFan, 2024-03-01:
A True dimension should have a 2d ref of the view (no geometry) + 3d refs (obj+subelement).
2 of your dimensions have 2d & 3d geometry references.   The third one is trying to be a length dimension of a face.  

You don't need to connect the 2d and 3d geometry.  The system will project the 3d points on to the 2d drawing for you.  Just select the view you want the dimension added to, plus some appropriate 3d geometry.

luzpaz, 2024-03-01:
@WandererFan verdict on this ticket ?

YosypC, 2024-03-03:
> A True dimension should have a 2d ref of the view (no geometry) + 3d refs (obj+subelement). 2 of your dimensions have 2d & 3d geometry references. The third one is trying to be a length dimension of [...]

I... didn't actually understand what you said. The dimension needs a 3D reference as it gives an error message ("True dimension has no 3D References), as the projected one clearly is not... well, true to the original dimension. Then it breaks.

Plus I didn't select any face, maybe you haven't seen well since there was a brief moment when it appeared I selected a face, but in reality it was an edge.

WandererFan, 2024-03-06:
I'm going by what is in your file.  The dimension has a reference of a Face.  That's never going to work for a length dimension.  

When you look at a dimension's references,  a projected dimension will have only 2d references based on points/edges on the page.  A true dimension will only have a 2d reference of the view (no geometry) and in the 3d reference section it will have object+subelement entries.

YosypC, 2024-03-06:
I might have missclicked in the file when I played around after the recording, but the videos clearly show multiple times that there's a bug by following the normal procedures. You can clearly see I select the edge and not the face.

WandererFan, 2024-06-10:
I think this was likely a devel code issue that has been fixed.

closed 2024-06-10 by WandererFan