FreeCAD/FreeCAD #12419: [Problem] Assembly : No way to explode the assemblies

PaddleStroke, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be good to have a way to explode the assembly.

My current idea is to have a button 'explode assembly' that would temporarily change the placement to explode the assembly. Then pressing the tool again (or cancelling the task) would put things back in place.
Thoughts?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-02-15:
Exploding an assembly isn’t just scattering the parts across the 3D space. It must be controlled in order to keep parts or sub-assemblies aligned. The explosion often needs to be partial. There should be the ability to have more than one exploded view (for assembly instructions). There should be lines to show where parts go (they must render in TechDraw). These lines shouldn’t be straight only.

The simple scattered expoded view is still useful so there could be two features (or sub-features): one for simple and usually temporary scattered view, one for more serious exploded views.

Depending on the strategy, it would be understandable to start small (scattered view) and to wait later for the more complex features.

Examples:

[image]

[image]

maxwxyz, 2024-02-16:
As an additional idea, what about temporarily also disable all joints in the exploded state and being able to move individual parts around by dragging. (if the part in the default exploded state hides something)
It must be also ensured when the auto explode is active, that no part is disappearing or intersecting with other parts (all parts should be visible).

As @pierreporte mentioned I'm also in favor of a simple auto explode and a more complex version. Maybe one can define the direction of the individual part and the distance / rotation how far this should move and then save this parameters for the exploded state. (e.g. create an assembly for the exploded state and save this settings in the assembly / joints)

PaddleStroke, 2024-02-27:
@qewer33 are you inspired to make an exploded view icon? :)

qewer33, 2024-02-27:
> @qewer33 are you inspired to make an exploded view icon? :)

I'll give it a shot! Any ideas?

maxwxyz, 2024-02-27:
What about the Assembly icon split apart?
@obelisk79 maybe interesting for DWG

qewer33, 2024-02-28:
A couple of proof of concepts:

[image]
[image]
[image]

[Assembly_ExplodedView.zip](https://github.com/FreeCAD/FreeCAD/files/14438333/Assembly_ExplodedView.zip)

obelisk79, 2024-02-28:
I like the yellow and blue cylinder concept.

maxwxyz, 2024-02-28:
> I like the yellow and blue cylinder concept.

But this could also be mistaken at small size for align axis or distance. Should we move this discussion to Discord?

pierreporte, 2024-02-28:
@qewer33 I prefer the last one. It uses the assembly container icons and explodes its 3 parts.

obelisk79, 2024-02-28:
That last one must have been an addtion... I like that the best so far

maxwxyz, 2024-02-28:
Same. Top lightning of the bottom yellow cube seems a bit too dark?
IMO you could remove the red lines as they are almost invisible at smaller sizes and depending on your theme.

pierreporte, 2024-02-28:
The red lines help understanding that the icon is for exploded views. They remain visible until the middle size, which is the one used for the toolbals. In the menu, the text helps further so it’s not a big deal if they are invisible.

PaddleStroke, 2024-02-29:
Let's go with the last one for now then

PaddleStroke, 2024-03-01:
@wwmayer I have a python binding question please. For this 'explode assembly view' tool I am trying to reuse the dragger (rather than reimplement the same thing in assembly).
Basically in this tool when I select a part I'm making a dragger appear on the part.
[image]

My problem is that I'm doing this in python : 

```
class ViewProviderExplodedView:
...
    def attach(self, vobj):
         ...
        self.dragger = coin.SoType.fromName("SoFCCSysDragger").createInstance()
        ...
```
But then the dragger object seems to not have all its c++ functions. For example `setUpAutoScale`.

So I am not sure how to handle this, because it seems that SoFCCSysDragger does not have a python interface. I can see different ideas : 
- Making a c++ class in assembly to interface the SoFCCSysDragger and make binding to this class.
- Make python interface for SoFCCSysDragger, but I don't know if it is possible ?
Thanks

pierreporte, 2024-03-01:
For an efficient workflow, the user must be able to align the dragger with part features. It was briefly discussed in #12451.

PaddleStroke, 2024-03-01:
Ah in fact it works it needs to be with * : 
`    return Base::Interpreter().createSWIGPointerObj("pivy.coin", "SoDragger *", asmDragger, 0);`

PaddleStroke, 2024-03-01:
Interesting, I can get the dragger object in python now, but trying to add the callback crashes freecad when the event triggers.
```
        self.asmDragger = self.assembly.ViewObject.getDragger()
        self.asmDragger.addFinishCallback(self.draggerMoved)
```
Interestingly I finally found that this problem was described 10 years ago : https://forum.freecad.org/viewtopic.php?style=10&t=5594
So there seems to be a workaround : 
```
view = FreeCADGui.ActiveDocument.ActiveView
view.addDraggerCallback(self.dragger, "addFinishCallback", self.move_dragger)
view.removeDraggerCallback(self.dragger, "addFinishCallback", self.move_dragger)
```

I will try that and feedback if it works.

PaddleStroke, 2024-03-01:
So adding the callbacks works.
```
        view.addDraggerCallback(self.asmDragger, "addFinishCallback", self.draggerFinished)
        view.addDraggerCallback(self.asmDragger, "addMotionCallback", self.draggerMoved)
```

But removing the callbacks crashes
```
        view.removeDraggerCallback(self.asmDragger, "addFinishCallback", self.draggerFinished)
        view.removeDraggerCallback(self.asmDragger, "addMotionCallback", self.draggerMoved)
```
It appears that SoFCCSysDragger is already adding a "addFinishCallback". So I'm adding a second one. And it appears `view.removeDraggerCallback` crashes in this case.

wwmayer, 2024-03-01:
I haven't tested it but very likely you have to do it this way:
```
cbFin = view.addDraggerCallback(self.asmDragger, "addFinishCallback", self.draggerFinished)
cbMov = view.addDraggerCallback(self.asmDragger, "addMotionCallback", self.draggerMoved)
```
```
view.removeDraggerCallback(self.asmDragger, "addFinishCallback", cbFin)
view.removeDraggerCallback(self.asmDragger, "addMotionCallback", cbMov)
```

PaddleStroke, 2024-03-02:
Thanks indeed it does not crash anymore!

closed 2024-05-06 by yorikvanhavre

FEA-eng, 2024-05-07:
This was closed by the large Assembly PR so there should be a tool for exploded views in Assembly now, right? I can't find it in weekly. Will it be added later?

maxwxyz, 2024-05-07:
I will add a PR to add the commands to the menu/toolbar

PaddleStroke, 2024-05-07:
The new tools are currently hidden by default. you need to set the bool parameter 'ExperimentalFeatures' in Mod/Assembly to true to make them appear.
We have done this so that the tools can be thoroughly tested by power users before they become shown by default.
So I would suggest that we let the situation like this for a few months to iron bugs out.

FEA-eng, 2024-05-07:
> So I would suggest that we let the situation like this for a few months to iron bugs out.

Ok, thanks for the information. But then they won't be available in 1.0 (feature freeze planned for the next week, release likely a few weeks later)?

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]