FreeCAD/FreeCAD #12420: [Problem] Arch Window is placed outside of selected face

quazgar, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For some walls, especially if they are created from faces and not by simply drawing each wall section by hand, window placement seems way off.

When placing a window, it sometimes is placed much closer to the camera or further away, than the selected face.  I think this is best illustrated by a screenshot:

[image]

### Full version info

```shell
Installed with Flatpak on Debian:

[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.1.23
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Here's a short screencast recording:

[window_in_wall.webm](https://github.com/FreeCAD/FreeCAD/assets/302587/2bbd75a3-745a-42ca-a1fe-9ae0c4405447)

Form posts like these seem to have similar problems, but no solutions:
- https://forum.freecad.org/viewtopic.php?style=4&t=7674

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-15:
Could you test this on the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roy-043, 2024-02-16:
This is a user error. See the documentation:

> When using presets, it is often convenient to turn the "Near" [Draft Snap](https://wiki.freecad.org/Draft_Snap) on, so you can snap your window to an existing face.

https://wiki.freecad.org/Arch_Window#Options

Alternative you can align the working plane to a face first.

Closing this.

closed 2024-02-16 by Roy-043