FreeCAD/FreeCAD #12424: [Problem] Default Windows Install is a UI Nightmare

maxwxyz, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The fresh install on a Windows machine is currently a UI mess. Text in the tree and on buttons are not readable, all dialog windows are crammed on the left side, dark overlays are applied, multiple half filled menu bars on the top reduce the working space in the 3D view.
I can see this drive away new users in seconds.
The default theme should be improved and the menu / first run wizard should only present working themes.
[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-02-15:
Added overlay for no_stylesheet.
Only need to change the yellow which is the default "objects being edited" color. 
[No_stylesheet_clean.zip](https://github.com/FreeCAD/FreeCAD/files/14302584/No_stylesheet_clean.zip)
Please test it, probably there are some issues that I have overlooked.

MisterMakerNL, 2024-02-17:
So here is the issue with this:
The default background color of the tree and the properties table is white.
Most background is white for default.
The background of the 3D view is far from white.
The overlay stylesheet overwrites the font colors.
A white font looks best on the 3D view background.
But you cannot even see it on the non-overlay background.
If you switch from overlay to non-overlay you cannot read it anymore.
If you make the text black you can read it very well in non-overlay.
But contrast is terrible in overlay.
One workaround is too use a fixed background.
But this overwrites the TreeEditColor, which is the color that highlights the active feature.
Another way is to use the effect but this makes a blurry background or weirdly shaped one.
Other way is too add a background on the frame, but this ruins overlay.

Ideal situation would be too switch stylesheets of a specific sidebar where overlay is turned off.
Not sure if that is possible.
Maybe we have to go for overlay on or off system wide, not per sidebar.
Or change the background color so we can use black text. But then we have an issue with the white sketch lines.
It all looks like a Mexican standoff.

MisterMakerNL, 2024-02-18:
So I got this working right, but not all too pretty.
But I set a priority for bugs instead of eye candy.
[image]

`* {
  /* color: #000000; */
  alternate-background-color: rgba(255, 255, 255, 0.437);
}

QTabWidget::pane {
  background-color: rgba(255, 255, 255, 0.566);
  border: transparent;
}

/* The OverlayTabWidget is named as OverlayLeft, OverlayRight, OverlayTop, OverlayBottom.
To customize for each overlay docking site, use the following selector

Gui--OverlayTabWidget#OverlayLeft {}
*/

Gui--OverlayTabWidget {
  qproperty-effectColor: rgba(255, 255, 255, 220);
  qproperty-effectBlurRadius: 5;
  qproperty-effectOffsetX: 0;
  qproperty-effectOffsetY: 0;
  qproperty-effectWidth: 2;
  qproperty-effectHeight: 2;
  qproperty-enableEffect: 1;
}
 Gui--OverlayTabWidget#OverlayBottom {
  qproperty-effectColor: rgba(255, 255, 255, 130);
  qproperty-effectBlurRadius: 10;
  qproperty-effectOffsetX: 0.0;
  qproperty-effectOffsetY: 0.0;
  qproperty-effectWidth: 2;
  qproperty-effectHeight: 2;
  qproperty-enableEffect: 1;
}

Gui--OverlayTabWidget::pane {
  background-color: transparent;
  border: transparent;
}

QSint--ActionGroup QFrame[class="content"] {
  border-bottom-left-radius: 5px;
  border-bottom-right-radius: 5px;
  }

Gui--OverlayTabWidget::tab-bar:top,
Gui--OverlayTabWidget::tab-bar:bottom {
    left: 10px;
    alignment: left;
}

Gui--OverlayTabWidget::tab-bar:left,
Gui--OverlayTabWidget::tab-bar:right {
    top: 10px;
    alignment: top;
}

Gui--OverlayProxyWidget {
  qproperty-hintColor: rgba(250, 250, 250, 0.6);
}

Gui--OverlayToolButton {
  background: transparent;
  padding: 0px;
  border: none;
}

Gui--OverlayToolButton::hover {
  background: rgba(250,250,250,200);
}

Gui--OverlayToolButton::focus {
  background: rgba(250,250,250,255);
}

Gui--OverlayToolButton::pressed,
Gui--OverlayToolButton::checked {
  background: rgba(150,150,50,80);
  border: 1px inset #f5f5f5;
}

Gui--OverlayToolButton::checked:hover {
  background: rgba(150,150,150,200);
  border: 1px inset #f5f5f5;
}

QTreeView{
  color: #000000;
}

QTreeView:disabled {
  color: #262626;
}

 QTreeView::item {
  /* border-radius: 3px; */
  margin: 1.5px 0px;
  /* background-color: rgba(255, 255, 255, 200); */
}

QFrame {
  border: none;
}

Gui--OverlayToolButton[objectName="OBTN Transparent"] {
  image: url(qss:overlay/transparent_dark.svg);
}

Gui--OverlayToolButton[objectName="OBTN Transparent"]:hover {
  image: url(qss:overlay/transparenthover_dark.svg);
}

Gui--OverlayToolButton[objectName="OBTN Overlay"] {
  image: url(qss:overlay/overlay_dark.svg);
}

Gui--OverlayToolButton[objectName="OBTN Overlay"]:hover {
  image: url(qss:overlay/overlayhover_dark.svg);
}

Gui--OverlayToolButton[objectName="OBTN AutoMode"] {
  image: url(qss:overlay/mode_dark.svg);
}

Gui--OverlayToolButton[objectName="OBTN AutoMode"]:hover {
  image: url(qss:overlay/modehover_dark.svg);
}

Gui--OverlaySplitterHandle Gui--OverlayToolButton,
Gui--OverlayToolButton[objectName="OBTN Float"] {
  image: url(qss:overlay/float_dark.svg);
}

Gui--OverlaySplitterHandle Gui--OverlayToolButton:hover,
Gui--OverlayToolButton[objectName="OBTN Float"]:hover {
  image: url(qss:overlay/floathover_dark.svg);
}`

MisterMakerNL, 2024-02-19:
Working on a fix here:
https://github.com/FreeCAD/FreeCAD/commit/872193230f23a64506cef90f043ec961712440b3
Currently issue with the mouse over function that doesn't work.

MisterMakerNL, 2024-02-27:
Also applicable here:
https://github.com/FreeCAD/FreeCAD/pull/11982#issuecomment-1967468104
If can apply a default dark stylesheet, I'd say move the others to the addon browser. Keep the standard ones included.

maxwxyz, 2024-04-06:
This following seems to be introduced after this issue was created.
Overlay panel titles are now displayed twice:
[image]

In general, the stable 0.21.2 is better than the current 0.22dev, see https://github.com/FreeCAD/FreeCAD/issues/13118
[image]

maxwxyz, 2024-06-28:
@MisterMakerNL I think I can close this as you've attacked most issues here.

closed 2024-06-28 by maxwxyz

MisterMakerNL, 2024-06-28:
Jup only bug hunting is left. Overlay is still a pain in the butt.