FreeCAD/FreeCAD #12425: Display, size and use of blue circles in Splines [Sketcher]

Tutorius, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new spline in sketcher, the set points are show by points surrounded by blue circles.
When zooming in, the circles get bigger and bigger. The mouse-pointer changes when getting near a blue circle, it changes to "PointOnObject". When clicking in such a case, the point is bound to the circles border.
After ending the edit, the control-points of this spline are bound together by constraints that are on invisible elements.

It would be nice if the size of the blue circles are scaled to a proportional value of the Zoom-Area.
It is of no need to be able to set constraints on these circles.

### Full version info

```shell
[code]
OS: Arch Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.5
  * ThreadProfile 1.89.0
  * fasteners 0.4.70
  * Nodes 0.1.36
  * Lithophane
  * sheetmetal 0.3.12
  * freecad.gears 1.0.0
  * Marz 0.0.27-alpha
  * Manipulator 1.5.0
  * A2plus 0.4.61
  * LCInterlocking
  * Silk 0.1.5
  * workfeature
  * lattice2 1.0.0
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * kicadStepUpMod 10.21.1
  * Render 2023.8.18
  * Curves 0.6.14
  * Help 1.0.3
  * MeshRemodel 1.8919.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Video of behaviour
https://youtu.be/_NjNsT5VR7g

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-15:
@Tutorius could you please test it on the current weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
There are some recent changes to splines and in the Sketcher WB in general.

Tutorius, 2024-02-16:
> @Tutorius could you please test it on the current weekly build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds There are some recent changes to splines and in the Sketcher WB in general.

Have tested this with the actual weekly build.
Its much better with changed view of circles by dashed lines to edit splines. The size of the circles still is not affected by zoom-size, so the cirlces get bigger, but they are smaller from the beginning.
It is still possible to fetch the circles for creating an "PointOnObject"-constraints.

There are plenty of shortkeys possible, they change to a change of mousecursor, but its not possible to add more points after that (a c d e h i kl l n o p s t v).
Using such a hotkey and pressing right mousebutton leaves the funtion, with no spine and all blue circles create before this visible and editable.

maxwxyz, 2024-02-16:
@PaddleStroke @AjinkyaDahale would it be possible to exclude the control point circles from auto snapping or change their size appearance (demo in the OP video link)

AjinkyaDahale, 2024-02-16:
I think this can be a part of bigger overhaul of the system: one where we can change the weights of the poles on creation.