FreeCAD/FreeCAD #12433: [Problem] MKS system is broken

Vanuan, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Toggle grid view.
Go to preferences -> General 
Change Unit system to MKS.
Hit apply. Update the view.

Expected: grid changes from millimeters to meters.
Actual: grid remains the same

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

The "Building Euro" unit system works though.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Vanuan, 2024-02-15:
Might be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/7395 and https://github.com/FreeCAD/FreeCAD/issues/5906

Roy-043, 2024-02-15:
Please try to be more specific.
Is this about the Draft grid?
What is the original unit system?
How should the grid change when the unit system changes?

maxwxyz, 2024-03-17:
@Vanuan could you add more info?

luzpaz, 2024-04-01:
Closing as duplicate.

closed 2024-04-01 by luzpaz