FreeCAD/FreeCAD #12436: In sketch the build order is fixed

flexbex, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Sketcher the build order in the Element list is fixed. This is important because tools like additive pipe and additive loft depend on the build order of sketch elements. So it would be great to be able to change the order via drag and drop.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Germany (en_DE)
Installed mods: 
  * kicadStepUpMod 10.19.5
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-20:
This feels like a duplicate.

maxwxyz, 2024-04-12:
I didn't find the duplicate.
To add from the FreeCAD Day 2024 Complaint Session:
**When extrusions between sketches the mapping of sketch vertices and other sketch vertices you cant individually chose the the vertex mapping**
Discussion: If you make multiple sketches and create a sweep, loft or a pipe it just makes crazy decisions. It's only twisting or whatever strange things that happen.
The connection between sketches / the mapping of sketch vertices - it's not possible to change for a user. You cannot choose how the sketches and features are connected or set the individual vertex mapping.

luzpaz, 2024-04-13:
ITC would you mind renaming it to a more descriptive title ?

Tryphis, 2024-08-19:
I don't know whether i have the same issue, but i got very inconsistent faces on a vertical sweep between two sketches. it seems to me like the enumeration of the relevant sketch points doesent work "correctly". This issue sounds related? I'm on 0.22.0dev. 
Cheers!
[image]