FreeCAD/FreeCAD #12438: Sketcher Tools Missing, Crashing

iAmVaeVictus, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Drawing a simple sketch on a datum plane, constrained from a spreadsheet. 
 
I'm not sure what led to this but it has happened today 2 times. Several of the Sketcher tool bars get closed, and I must reopen them all. 

The only tool bars shown are the ones with V's in front of them below: 
V File
V Edit
Clipboard 
v Workbench 
v Macro
V View 
v Help
V Auto-Created Macro Toolbar
Sketcher edit mode
Sketcher geometries
Sketcher constraints
Sketcher tools
Sketcher B-spline tools
Sketcher virtual space
Sketcher edit tools

1. Open FC. Open Sketch from Tree view (SketchScrewFlapBase)
2. Toolsbars are present. 
3. do some work, open spreadsheet, tile the spreadsheet and sketch
4. Touch the sketch window
5. Note that toolbars are closed. 
6. Hit Save
7. Click Workbench dropdown (Maybe that's what I did, not sure - I clicked something up there)
8. Crash
9. Reopen FC
10. open Sketch, toolbars still gone
11. Reopen tool bars for sketch. 
12. Try to recreate crash or missing toolbars.... nothing yet but it happened twice now. 

Apple Crash report and project file attached. 

### Full version info

```shell
[code]
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[FC SketcherToolbarsGone AppleCrash Report.txt](https://github.com/FreeCAD/FreeCAD/files/14306085/FC.SketcherToolbarsGone.AppleCrash.Report.txt)
[Switch Housing v7.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14306097/Switch.Housing.v7.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-02-16:
I could not reproduce on

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35992 (Git)
Build type: Debug
Branch: main
Hash: 43723b6d780fba4d6f359836842cd7f6cb51c3e5
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

The crash report says that this is really close to this [^1].
[^1]: https://github.com/FreeCAD/FreeCAD/issues/9198#issuecomment-1763448152

maxwxyz, 2024-02-16:
I cannot reproduce as well.
@iAmVaeVictus can you try it with the current dev builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

iAmVaeVictus, 2024-02-17:
Further info on this: The tools disappearing seems to happen when I 

1) have a spreadsheet and a part view tiled 
2) bounce between editing a sketch, with the Task tab in focus, and clicking the Model view to turn on body parts, then instead of clicking the Task tab again, double clicking on the Sketch in the tree that is already open in the Task tab. I *think* that is when the tools go away. 
3) this does NOT create a crash however....

iAmVaeVictus, 2024-02-17:
And Yes, I will download and try the new builds!

maxwxyz, 2024-02-19:
Closing - not reproducible in dev builds

closed 2024-02-19 by maxwxyz

RicoChr, 2026-04-10:
I am experiencing a similar issue in release 1.1.0 on MacOS. 

### Steps to reproduce
Open a sketch - all toolbars active by default are active.
Make a different body active, open a sketch - all toolbars are still active.
Activate the first body again, open a sketch - all sketch toolbars are disabled.

maxwxyz, 2026-04-10:
@RicoChr can you reproduce in the last weekly 1.2?