FreeCAD/FreeCAD #12440: [Problem] Assembly: Nested Assemblies (ASM in ASM)

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is currently not possible to have nested assemblies / assemblies inside an assembly.
This is crucial for complex models or if you are collaborating with other persons/teams so each one can work on different aspects of the entire larger assembly. 
Also to assembly parts (Part Container) consisting of multiple bodies in a parent assembly.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-11:
I guess B would be better, otherwise free movement of not constraint directions would not be possible if they would be handled as a sloid (option A). This would be needed for interaction, animation, etc. later on. 
Not sure about pulling the child joints to the main assembly. Couldn't they be inside the sub-assembly? What would happen if you pull in an assembly file (or just link one) as a sub-assembly in a current assembly and the joints move outside? Will the sub-assembly file still work on its own (like a normal assembly)?

PaddleStroke, 2024-03-11:
I didn't meant to actually move the joint objects. Just when you solve an assembly you get all the joints of the assembly + the joints of the child asembly to solve it

pierreporte, 2024-03-11:
@PaddleStroke In CAD program you usually have the choice between A and B. A is the default: parts in an assembly act like a rigid part when manipulated from outside the assembly. B is available as an option (in Solidworks it is a flexible assembly) and is used for pneumatic cylinders for example.

For motion simulations, I believe that assemblies should be considered as B regardless of what they actually are. Altough I don’t have experience with such things.

tobiasfalk, 2024-03-11:
B sounds like a good default solution, but I think having the option to fix/freez/solidefy/"disable solving" a Assembly could be useful for multiple reasons.
One would be to move the subassembly in space without weird behaviour, for this usecase the ability to set a shortcut to it would be nice(like holdding 'F' while draging).
An other could be to improve solving time when doing a large Assembly, as in selectively disabling single Assembly's that are currently not needed, that is if it improves solving time.
But ther definitely will be more.

PaddleStroke, 2024-03-12:
For now I have done B. If we want to enable A as well it will need more thoughts about how to switch from one to another. Anyway static sub-assemblies can be done with App::Parts. But then there is no going back and forth freezing and unfreezing.

Now I have added support but only for Assemblies. Not for links to assemblies.
It is going to be tricky to support links to assemblies if we support B because if you have 2 links to the same sub assemblies, then the objects within can't move differently.

PaddleStroke, 2024-03-12:
One solution I see about this, is that the insert tool handles assembly in a different way: 
Instead of creating a link object to the assembly, it would create an assembly and populate it with links to each objects of the assembly.

tobiasfalk, 2024-03-12:
> Instead of creating a link object to the assembly, it would create an assembly and populate it with links to each objects of the assembly.
Sounds good, but what happens when are adds a part to the "original" assembly, will it also be added to the subassembly?

pierreporte, 2024-03-12:
Static assembles are not always static in a mechanical point of view. Even then, it can change. As you say, it is impossible to switch from Part to Assembly so it would be dangerous to use the Part container for this.Technically flexible assemblies are sufficient because constrains should be used to fix parts together. However it works rarely like this in the industry where there are often joints missing or in error. The biggest advantage of fixed assemblies may be that they may not require complex recomputations each time they are touched.From experience, flexible assemblies are very rarely needed unless running a motion simulation, in which case they should all turn flexible temporarily, and the positions + static status should go back to what they were.I think that static assemblies should be default but there could be an option for the user to specify what should be done.

PaddleStroke, 2024-03-12:
No with this idea the new assembly will not be connected to the 'original'. So it's not ideal.

But other than that I see no solution to support more than 1 link to the same assembly.

tobiasfalk, 2024-03-12:
> No with this idea the new assembly will not be connected to the 'original'. So it's not ideal.
> 
> But other than that I see no solution to support more than 1 link to the same assembly.

Could a check for difference be done when recomputed? Maybe even as an button, update subassemblys.

tobiasfalk, 2024-03-12:
What I mean is that, the subassembly knows from where it originated from and when the update event is called it looks if there is any difference between it amd the origin

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre