FreeCAD/FreeCAD #12441: [Problem] Assembly: Selecting an Element of a Body does not select it in the Tree

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Selecting an element of a part/body inside of an active assembly does not select the corresponding body in the tree view, which makes it hard to see what is currently selected. (Sync selection setting)
Very hard for imported STEP files, where each body is a blank solid, sometimes without a proper name.
You have to right click on the object and click on go to selection to see the selection updated in the tree.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-03-21:
This is not an assembly issue but a partdesign issue. This behavior does not depend on the body being in an assembly or not.

maxwxyz, 2024-03-21:
@PaddleStroke but its not a PartDesign container, also for imported STEP solids (without additional PD features)

PaddleStroke, 2024-03-22:
But this behavior of imported step solids, does it not also happen outside of an assembly container?

maxwxyz, 2024-03-22:
I cannot reproduce with the current weekly

closed 2024-03-22 by maxwxyz