FreeCAD/FreeCAD #12445: [Problem] Assembly: Surface Contact / LCS Constraint not possible

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to select a non planar surface for a contact constraint, because the LCS is generated at the center of the surface.
It would be ideal if the surface itself can be used for joints (e.g. distance to another surface) an also for collision prevention / detection.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-16:
Actually the distance joint does not care about the JCS. It only cares about the selected element.
So if you select a cylindrical face and a plane, you will get distance between them (or tangent if distance is 0).

Having said that I have just tried to do what I just said and notice 2 things : 
- It does not work. It makes a plane to plane distance using the JCS.
- The JCS could be hidden when we make a distance joint.

maxwxyz, 2024-02-16:
@PaddleStroke also, the offset in the distance joint is not displayed directly in the 3d view when you change it, but you have to click ok to see it move to the right location. All other tools directly show the output of the offset

maxwxyz, 2024-02-16:
@PaddleStroke I've updated the original post with more issues of the distance joint and two demo videos.

PaddleStroke, 2024-02-16:
Thanks.
So there are several things : 
- I implemented a 'pre-solve' that matches the JCS. This was to avoid wrong solutions from the solver. However this pre-solve does not make sense for distance joints. Especially not for cylinder cases.
- Cone surfaces distances are not defined because we don't know how to handle these with the solver.
- In your video the ball-cylinder joint and ball-plane distances seems to work as expected (after creation).

So overall I need to rework the pre-solve to accomodate distance joints

maxwxyz, 2024-02-16:
Torus (donut) surfaces or spline based / irregular surfaces also do not seem to work. I get it that it is hard to calculate them.

PaddleStroke, 2024-03-11:
I think this issue is resolved by https://github.com/FreeCAD/FreeCAD/pull/12406, but if you can check the PR to see if all bug fixed ? Because this issue seems to bundle many things so I'm not sure.

maxwxyz, 2024-03-11:
@PaddleStroke 

2. fixed
3. fixed

1/4 seems too hard to fix.

closed 2024-05-06 by yorikvanhavre

pierreporte, 2024-05-11:
@maxwxyz Should new issues be created for the remaining problems? I was trying to make two conical surfaces in contact (for screws).

maxwxyz, 2024-05-11:
@pierreporte yes please, if it is solvable.

pierreporte, 2024-05-11:
@maxwxyz Done with #13958. It needs fixing in any case, the operation is very common.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]