FreeCAD/FreeCAD #12448: [Problem] FEM CalculiX execute error with modified inp file

FEA-eng, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Download this file and change the extension to .FCStd: [Execute error.zip](https://github.com/FreeCAD/FreeCAD/files/14310399/Execute.error.zip)
2. Open it in the newest 0.22 weekly build (it uses the newly implemented FEM features)
3. Go to the FEM workbench, double-click on Analysis to activate it. then double-click on SolverCcxTools.
4. Click Write .inp file, then wait and click Edit .inp file
5. Scroll down and replace this:

```
** Tie Constraints
** ConstraintTie
*TIE, POSITION TOLERANCE=0.1, CYCLIC SYMMETRY, NAME=TIEConstraintTie
TIE_DEPConstraintTie, TIE_INDConstraintTie
*CYCLIC SYMMETRY MODEL, N=8, NGRAPH=8, TIE=TIEConstraintTie, ELSET=Eall
0,0,0,0,0,1

***********************************************************
** At least one step is needed to run an CalculiX analysis of FreeCAD
*STEP
*STATIC

***********************************************************
** Fixed Constraints
** ConstraintFixed
*BOUNDARY
ConstraintFixed,1
ConstraintFixed,2
ConstraintFixed,3

***********************************************************
** Displacement constraint applied
** ConstraintDisplacement
*BOUNDARY
ConstraintDisplacement,1,1,0.0

***********************************************************
** Outputs --> frd file
*NODE FILE
U
*EL FILE
S, E
** outputs --> dat file
** reaction forces for Constraint fixed
*NODE PRINT, NSET=ConstraintFixed, TOTALS=ONLY
RF
** reaction forces for Constraint displacement constraining translation
*NODE PRINT, NSET=ConstraintDisplacement, TOTALS=ONLY
RF

***********************************************************
*END STEP 

***********************************************************
** CalculiX Input file
**   written by    --> FreeCAD 0.22.0
**   written on    --> Fri Feb 16 12:35:23 2024
**   file name     --> Execute error.FCStd
**   analysis name --> Analysis
**
**
***********************************************************
**  About units:
**  See ccx manual, ccx does not know about any unit.
**  Golden rule: The user must make sure that the numbers they provide have consistent units.
**  The user is the FreeCAD calculix writer module ;-)
**
**  The unit system which is used at Guido Dhondt's company: mm, N, s, K
**  Since Length and Mass are connected by Force, if Length is mm the Mass is in t to get N
**  The following units are used to write to inp file:
**
**  Length: mm (this includes the mesh geometry)
**  Mass: t
**  TimeSpan: s
**  Temperature: K
**
**  This leads to:
**  Force: N
**  Pressure: N/mm^2 == MPa (Young's Modulus has unit Pressure)
**  Density: t/mm^3
**  Gravity: mm/s^2
**  Thermal conductivity: t*mm/K/s^3 == as W/m/K == kW/mm/K
**  Specific Heat: mm^2/s^2/K = J/kg/K == kJ/t/K
**
```

with this:

```
** Tie Constraints
** ConstraintTie
*TIE, POSITION TOLERANCE=0.1, CYCLIC SYMMETRY, NAME=TIEConstraintTie
TIE_DEPConstraintTie, TIE_INDConstraintTie
*CYCLIC SYMMETRY MODEL, N=8, NGRAPH=8, TIE=TIEConstraintTie, ELSET=Eall
0,0,0,0,0,1

*NODE
20000,0,0,200
20001,0,0,200
*RIGID BODY, NSET=ConstraintDisplacement, REF NODE=20000, ROT NODE=20001
***********************************************************
** At least one step is needed to run an CalculiX analysis of FreeCAD
*STEP
*STATIC

***********************************************************
** Fixed Constraints
** ConstraintFixed
*BOUNDARY
ConstraintFixed,1
ConstraintFixed,2
ConstraintFixed,3

***********************************************************
** Displacement constraint applied
** ConstraintDisplacement
*CLOAD
20001,3,1000

***********************************************************
** Outputs --> frd file
*NODE FILE
U
*EL FILE
S, E
** outputs --> dat file
** reaction forces for Constraint fixed
*NODE PRINT, NSET=ConstraintFixed, TOTALS=ONLY
RF
** reaction forces for Constraint displacement constraining translation
*NODE PRINT, NSET=ConstraintDisplacement, TOTALS=ONLY
RF

***********************************************************
*END STEP 

***********************************************************
** CalculiX Input file
**   written by    --> FreeCAD 0.22.0
**   written on    --> Fri Feb 16 12:25:03 2024
**   file name     --> 
**   analysis name --> Analysis
**
**
***********************************************************
**  About units:
**  See ccx manual, ccx does not know about any unit.
**  Golden rule: The user must make sure that the numbers they provide have consistent units.
**  The user is the FreeCAD calculix writer module ;-)
**
**  The unit system which is used at Guido Dhondt's company: mm, N, s, K
**  Since Length and Mass are connected by Force, if Length is mm the Mass is in t to get N
**  The following units are used to write to inp file:
**
**  Length: mm (this includes the mesh geometry)
**  Mass: t
**  TimeSpan: s
**  Temperature: K
**
**  This leads to:
**  Force: N
**  Pressure: N/mm^2 == MPa (Young's Modulus has unit Pressure)
**  Density: t/mm^3
**  Gravity: mm/s^2
**  Thermal conductivity: t*mm/K/s^3 == as W/m/K == kW/mm/K
**  Specific Heat: mm^2/s^2/K = J/kg/K == kJ/t/K
**
```

6. Click Save, then Run CalculiX. You will get the following error:

`CalculiX execute error: PySide2.QtCore.QProcess.ProcessError.Crashed`

The .inp file itself is correct. It uses proper CalculiX syntax and works when submitted in standalone CalculiX outside of FreeCAD. So no idea what's going on here.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-16:
Is this a FreeCAD bug or regarding calculix? Does it work when used without FreeCAD?

FEA-eng, 2024-02-16:
> Is this a FreeCAD bug or regarding calculix? Does it work when used without FreeCAD?

I've mentioned this at the bottom of the issue description - it works in CalculiX itself so it must be a FreeCAD bug. Not even an issue with the ccx binary provided with weekly builds because it works when I submit the input file using that binary. So something must be wrong with the execution procedure in FreeCAD.

maxwxyz, 2024-02-16:
I'm not familiar with ccx: is this a normal thing to do in FEM WB to adjust the inp file? Would you consider this a bug or a new feature to be implemented?

marioalexis84, 2024-02-16:
@FEA-eng I can't reproduce this.
The error message should only appear if CalculiX crash or if you stop the process with the stop button.

FEA-eng, 2024-02-17:
> @FEA-eng I can't reproduce this.

On what OS ? I can't reproduce it on Ubuntu now so maybe it's Windows-specific. I will have to try on other computers with this OS.

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git) AppImage
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

marioalexis84, 2024-02-17:
On Linux.
Can you reproduce with another example?

FEA-eng, 2024-02-17:
I've tried on a different Windows computer and the same error occured. Can someone with Windows confirm ?

I will let you know if I find another example.

marioalexis84, 2024-02-17:
> Can it be caused by https://github.com/FreeCAD/FreeCAD/pull/12273 ?

It's the most likely. This is clarified in the Qt documentation in relation to a signal delay:

>The [QProcess](https://doc.qt.io/qt-6/qprocess.html) object will immediately enter the Starting state. If the process starts successfully, [QProcess](https://doc.qt.io/qt-6/qprocess.html) will emit [started](https://doc.qt.io/qt-6/qprocess.html#started)(); otherwise, [errorOccurred](https://doc.qt.io/qt-6/qprocess.html#errorOccurred)() will be emitted. Do note that on platforms that are able to start child processes synchronously (notably Windows), those signals will be emitted before this function returns and this [QProcess](https://doc.qt.io/qt-6/qprocess.html) object will transition to either [QProcess::Running](https://doc.qt.io/qt-6/qprocess.html#ProcessState-enum) or [QProcess::NotRunning](https://doc.qt.io/qt-6/qprocess.html#ProcessState-enum) state, respectively. On others paltforms, the [started](https://doc.qt.io/qt-6/qprocess.html#started)() and [errorOccurred](https://doc.qt.io/qt-6/qprocess.html#errorOccurred)() signals will be delayed.

I will do the test from a virtual machine with Windows to correct this problem.

marioalexis84, 2024-02-17:
@FEA-eng can you share the complete output you get until the error message is printed?

FEA-eng, 2024-02-17:
> @FEA-eng can you share the complete output you get until the error message is printed?

Sure:

```
0.0: Check dependencies...
0.1: Write completed.
0.0: CalculiX binary: E:/FreeCAD_weekly-builds-35994-2024-02-16-conda-Windows-x86_64-py310/bin/ccx.exe
0.0: CalculiX input file: C:\Users\NJ\AppData\Local\Temp\fcfem_gc85zoii\FEMMeshGmsh.inp
0.0: Run CalculiX...
0.0: Starting CalculiX...
0.0: CalculiX is running...
0.0: CalculiX execute error: PySide2.QtCore.QProcess.ProcessError.Crashed
0.0: CalculiX stopped.
```

marioalexis84, 2024-02-17:
@FEA-eng I sent to you a message in the forum,

commit 38fdba4a (2024-02-19): Fem: Prevent stop CalculiX immediately - fixes #12448
closed 2024-02-19 by chennes