FreeCAD/FreeCAD #12455: [Problem] Assembly: Animate Assemblies

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It should be possible to animate the assembly. Asembly animations facilitate communication and documentation by visually illustrating the assembly process and component relationships.

It should be possible to define motion types for parts or joints incl. start/end position and speed. Define the motion relation between parts.
Play the animation and export it.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-15:
This is at the junction of assembly and MbD. This is the next major step for assembly workbench imo (after BoM which is a smaller project).
José has a Motion addon workbench, I hope we will be able to integrate it step by step in integrated assembly.

luzpaz, 2024-08-27:
Is this just available in Ondsel build atm ?

PaddleStroke, 2024-08-28:
Not yet no.

Zolko-123, 2024-09-11:
> Is this just available in Ondsel build atm ?

It's available in Assembly4 + FreeCAD since years

luzpaz, 2024-09-11:
> It's available in Assembly4 + FreeCAD since years  

True and i was aware of that. But thanks for the reminder. I've added a label for posterity.

Zolko-123, 2024-09-13:
> for posterity

that's a funny one, might come back to bite you

luzpaz, 2024-09-13:
It seems the biting has already begun ?

Zolko-123, 2024-09-13:
History will tell

luzpaz, 2024-09-13:
By all means, if you have something to say, please elaborate what your concerns are.

Zolko-123, 2024-09-13:
you mean apart from all the re-inventing and re-implementing of existing features in new and incompatible ways ?

maxwxyz, 2025-01-22:
@PaddleStroke should be implemented