FreeCAD/FreeCAD #12456: [Problem] Assembly: BOM export

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no way to export a BOM (bill of materials) from an assembly. It would be great to generate a spreadsheet with the part name, amount, unit, and more part specific details (if they can be defined; e.g. weight, size,..) and custom parameters like part numbers.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-02-19:
See Paul's WIP workbench:

[Bill Of Materails FreeCAD WB](https://github.com/APEbbers/BillOfMaterials-WB)
[Post in the forum](https://forum.freecad.org/viewtopic.php?p=738229)

PaddleStroke, 2024-03-21:
https://github.com/APEbbers @APEbbers

PaddleStroke, 2024-03-21:
@APEbbers would you consider making a PR to main freecad to integrate your bom tool to the assembly workbench?

PaddleStroke, 2024-03-21:
@pierreporte @maxwxyz can you please checkout Paul's BOM functionality and make any comments about functionality issues or such? So that if we get it in core we can get it right.

pierreporte, 2024-03-24:
@PaddleStroke Overall the base features are there. The major issues are the first three. The other ones could be considered refinements even though it’s far from trivial. Besides this, the list is not in a particular order.

-----------------------------

Sensitive to #12907 (the lock symbol is present in the spreadsheet).

Available columns not very useful, but this is actually #12136.

Sensitive to #11742 because it asks for bodies when a BOM should be about parts only.

When a new BOM is created, the spreadsheet object is replaced so manually entered data is lost.

Number column is a hierarchical value (1 for root, 1.1 for first child, 1.2 for the second, etc.) It can be useful but, if it is to be linked with a drawing, it needs to be only integers. The column itself may not be useful by default.

Unnecessary styling by default.

The file information block should be on top to comply with CSV format: tt should be in the header.

Same for summary block.

`CreateBOM_Raw` adds bodies even when told not to.

BOM doesn’t update automatically when new parts are added.

I think that it shouldn’t work with spreadsheets, except for pre-export so the user can do things before exporting to CSV, Excel or whatever. It should be a window with all the options so it changes dynamically depending on what the user selects (single level, raw, columns, etc.) This window would have buttons to create a spreadsheet object or export directly to a file. The window would be a good basis for future mass analysis of the assembly (to see what part is the heavier).

The UI/UX could be way better even without that window: there should be just one button. The task box would give all the options for every possible scenario. The Create BOM button should be on top of the task panel like for sketches.

----------------------------

Maybe take inspiration from Alibre:

* https://help.alibre.com/articles/#!alibre-help-v27/bill-of-materials-boms
* https://www.alibre.com/fr/bill-of-materials/

pierreporte, 2024-03-24:
I can’t create a BOM if there is already a TechDraw page in the document.

When I insert a spreadsheet view of the BOM, it shows the following first picture instead of the second:

[image]
[image]

A separate spreadsheet object is useless for drawing. I get that it is necessary for the current implementation.

This is the assembly used: [bom test.zip](https://github.com/FreeCAD/FreeCAD/files/14735187/bom.test.zip) (contains multiple FCStd files)

snelweg, 2024-06-04:
Have been using the asm_bom branch for a little while and it looks very promising.
Will it be possible to extend the bom functionality so that we can make cutlists?

PaddleStroke, 2024-06-04:
thanks for the test. Yes it should be fairly easy to expand the functionality. By cutlists you mean to be able to select objects to exclude?

APEbbers, 2024-06-04:
I think you mean a list of profiles (Beams, C-profiles, tubes, etc.) with the length and angle they need to be cut to.
It is often used with weldments, and construction work.

@snelweg: Am I right?

snelweg, 2024-06-04:
> I think you mean a list of profiles (Beams, C-profiles, tubes, etc.) with the length and angle they need to be cut to. It is often used with weldments, and construction work.
> 
> @snelweg: Am I right?

Almost.
I should have mentioned I am a woodworker.

We woodworkers need slightly  more information then if it was just a bunch of weldments. Wood-panels and beams do not have standardized measurements as steel tubes and such have.
We want the length, width en depth (woodworkers tend to make body's from a sketch and pad that sketch). 

In OnShape there is a cutlist, but of course it only works for steel beams. I tried to use that metal cutlist but it just does not work with wood.

There are some attempts at proper woodworking BOM;s and cutlist's in FreeCAD. The one in Linkstage3 is very good. The one in Assembly4 wb is also working and is expandable.

With a cutlist we will be able to use online nesting software and calculate what boards and panels we have to buy.

snelweg, 2024-06-04:
> thanks for the test. Yes it should be fairly easy to expand the functionality. By cutlists you mean to be able to select objects to exclude?

No. Woodworkers use the dimensions of the panels from a BOM to generate a cutlist. The cutlist shows all the same materials grouped together (ie thickness).  We can use that list for making a nested view of all the materials we need for a project.

It's very well explained here: https://woodbin.com/ref/project-design-and-planning/bill-of-materials/

Some on-line nesting programs that you feed a cutlist to and it generates nice view of complete panels with all the cuts you have to make.

https://www.cutlistoptimizer.com/

https://www.opticutter.com/linear-cut-list-calculator

Thanks for starting the BOM, I hope there will be some cooperation with all the other people that are working on that functionality.

snelweg, 2024-06-04:
I found something that works for the way most woodworkers model their projects.
https://github.com/dprojects/getDimensions/tree/master/Docs

PaddleStroke, 2024-06-05:
Ok so I understand the need. Yes the current tool can be expanded to do this. We could add 'length, width, thickness' columns automatically generated from the solid dimensions. And add a way for user to sort objects by a property.

snelweg, 2024-06-05:
> Ok so I understand the need. Yes the current tool can be expanded to do this. We could add 'length, width, thickness' columns automatically generated from the solid dimensions. And add a way for user to sort objects by a property.

Great, that is good news.

Where I said bodies in my initial response I mean parts btw. The parts are pieces of wood which have been worked on, they are final parts and are used in an assembly.

There are many things to consider to make a good BOM. It's good to see the basics are there already.

[image]

snelweg, 2024-06-06:
As a follow up.
In the early days I used OpenBom when it was free and nice. Now it is very expensive and I will not use it anymore.
Luckely there is IndaBom now, looks promising. Hope there will be a connector for them so we can use the BOM from Ondsel inn there.
This would be of benefit to all the users, not just the woodworkers.

This will introduce a naming scheme that is used in most industries along the lines of:

> Part numbers must be in the format CCC-NNNN-VV where CCC is the part number class, NNNN is the part number number item, and VV is the part variation.

https://indabom.com/bom/help/

pierreporte, 2024-06-06:
@snelweg It is not a good idea to tie a BOM/PDM/PLM solution with FreeCAD at this level. It should be some sort of API with add-ons provided by vendors, similar to Ondsel Lens. PDM/PLM support is absolutely essential, and it is required in some industries, so it should be possible to plug any software to FreeCAD with an experience as close as native, not just one.

PR #14198 (2024-06-17): Assembly: Implement Bill Of Materials
Implement BOM object / tool. Fixes #12456

[image]
closed 2024-06-17 by chennes