FreeCAD/FreeCAD #12458: [Problem] Sketcher: Sketch plane orientation is unclear (no axis cross)

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The orientation of a sketch is unclear in the 3D space. If the sketch is not active (= just visible in the 3d space, e.g. in PartDesign) the origin and the horizontal and vertical axis (H/V) incl. direction is not visible. This also applies for datum planes.
When the sketch is edited, the differentiation is only from the axis color, but you cannot see which direction is positive H or positive V when not looking directly at the sketch plane (camera aligned to normal plane vector).
Also when in the attachment editor, the origin and axis cross for a sketch plane is not visible. It is mostly guessing and trial and error what you click.

The proposal is to include an option to display an axis cross at the origin of the sketch, with arrows indicating positive H and positive V direction. If enabled, the axis cross should be displayed when the sketch is active (in edit mode) and also when the sketch is visible and not active.

Here is an example from CATIA:
[image]

This allows to quickly understand the location and orientation of the sketch plane.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/FreeCAD/FreeCAD/issues/11781#issuecomment-1868249260

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-13:
To add from the FreeCAD Day 2024 Complaint Session:
**Selecting coordinate system for a sketch is convoluted, there are multiple ways to do it, when editing a sketch the coordinate system is not visible**
This issue is linked to the second mention.
Discussion:
When you enter the sketch it is also absurd. There is a feature where you cut out all the stuff that is in front which could be enabled by default. Once you're actually editing the sketch, the sketch plane is not visible because it may be obscured.

Related:
- https://github.com/FreeCAD/FreeCAD/issues/12584
- https://github.com/FreeCAD/FreeCAD/issues/11133

pierreporte, 2024-04-21:
I agree with @maxwxyz. Unlike Catia, I wouldn’t delete the two main axis but rather transform them into construction line (same color for both). The arrows with their label would be on top on them. The construction point at the sketch origin would remain as well. The labels could be H/V or X/Y.

[image]

From outside the sketch, a LCS could be shown at the origin. By default, it could be hidden except when the sketch is selected. The origin of the sketch should be visible and pickable when the LCS is shown, but it should remain a construction point from inside the sketch.

[image]