FreeCAD/FreeCAD #12459: Path Dressup Tag has multiple errors

acp693, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Almost impossible to place a tag manually. Often the path generated when viewed from the top impinges into the model which would be catastrophic for the piece being milled. Or a proper tag isn't generated which would result in the tag being milled off. The following errors are common when placing tags manually:

Tags.ERROR: Exception during processing tag @(269.87, 83.97) (Both points are equal) - disabling the tag

Tags.ERROR: Exception during processing tag @(90.19, -73.36) (No connection to Vector (90.19480160616936, -73.35932385761461, 3.99959837273243)) - disabling the tag

Even the default settings on opening the dressup Tag dialog sometimes generates errors when using the automatically generated tag placement function and one or two tags will sometimes be disabled, especially if the model isn't straight sided but is instead composed of curves. If the default number of automatically generated tags is increased, they are not distributed evenly around the model.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.4
  * Curves 0.5.3
  * ExtremeProLight 1.4.3
  * FeedsAndSpeeds 0.5.0
  * Manipulator 1.5.0
  * Marz 0.0.26-alpha
  * Silk 1.0.0
[/code]
```

### Subproject(s) affected?

Path

### Anything else?

[Tags.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/14313904/Tags.FCStd.gz)
[Tags2.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/14313907/Tags2.FCStd.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-16:
@acp693 Could you try it in the weekly builds. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
The PathWB had some improvements in development.

acp693, 2024-02-16:
@maxwxyz I just tried it in the latest windows dev build, it's the same behaviour,  placing a tag manually nearly always fails completely, or fails to build a path around the tag position, but in the dev version, unlike the stable version, there's no way to recover from it, the dialog becomes greyed out including the 'Save' and 'Cancel' buttons, leaving no option but to force quit freecad. 

One thing I've noticed in the dialog for adding tags manually is that there are three input boxes for X,Y, and Z, Z is greyed out and set to 0,000. However, after placing a tag, the Z box often shows a different value other than zero, some small value. Perhaps this has some baring on the bug?

ninao-xyz, 2024-05-07:
I've tested this on the latest weekly builds. The issue is still there. It's practically impossible to use the TagDressup when milling multiple parts in the same job.

here's the error I get in the offending tags:
`
02:47:13  Tags.ERROR: Exception during processing tag @(117.00, 528.84) (Both points are equal) - disabling the tag`

This is a really blocking bug for people that use FreeCad CAM for milling wood parts. I hope someone can take a look into it before the next release.  I've spend 2 hours in a project trying to make the tagsDressup work without success. (Not trying to bash anyone, bugs append) Right now I'm exporting my models into blenderCAM to mill them as a workaround.

FreeCad info:
```
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.22
  * OpenDark 2023.12.17
  * SelectorToolbar
  * FeedsAndSpeeds 0.5.0
  * CubeMenu
```

ninao-xyz, 2024-05-10:
Done more digging and I've noted that some tags are affecting adjacent paths instead of just the one it is supposed to.
Uploaded a screenshot to better describe what I'm trying to tell. 
[image]

```
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.22
  * OpenDark 2023.12.17
  * SelectorToolbar
  * FeedsAndSpeeds 0.5.0
  * CubeMenu
```

mrnro, 2024-06-02:
I had the same issue on a circular path and found out that changing the angle from default 45 degrees to 90 degrees let me add the tags manually:

[image]
[image]

`
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
`

acp693, 2024-06-05:
@mrnro I cannot confirm your findings using the model I attached to this bug report. I tried 90° and didn't find it any easier to add a tag. One thing that is different with using 90° is that when a tag is generated, it no longer sometimes generates a path that sometimes impinges into the model as often happens when using the 45° angle. On further investigation, any angle above 45° avoids this serious problem (at least the handful of angles I tried). Look at the attached version of the model (Tags3). Open the Tag dialog, and look at the model from the top, one tag is generating a path that cuts into the model. If you set the angle to 46° this no longer happens.

One thing I've found that helps adding a tag is zooming in massively onto the path where the tag is to be attached so that the mouser pointer is exactly on the path line, don't click the mouse, instead tab through to highlight the 'Save' button and press return. Even then adding the tag quite often fails. (this is mainly on curved paths as far as I can see, straight sided paths seem to be unproblematic)

One other thing, if I click on a tag in the model view so that it is highlighted, I can't press delete on the keyboard to delete it, I have to click the delete button in the Tag dialog box.

[Tags3.FCStd.zip](https://github.com/user-attachments/files/15594475/Tags3.FCStd.zip)

OS: Windows 10 build 19045
Word size of FreeCAD 64-bit
Version:   0.22.0dev.37645 (git) 
Build type: Release
Branch: Main
Python: 3.11.9, qt 5.15.13, coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

JohnOCFII, 2025-02-21:
Pretty sure this is still a valid issue.

cdsdonham, 2026-01-03:
I believe I encountered this issue today (1/3/26).  I poked at the code some and noticed the following:  In CAM/Path/Dressup/Tags.py, there appears to be a magic distance coded into pointIsOnPath.  It is hardcoded to allow a maximum of 0.1 in a call to Path.Geom.isRoughly, but for my clicks, the distance is 0.12.  This causes the click to not be allowed and the tag not to be created.  Simply changing the value to 0.15 gets beyond this, but does not solve the problem as the point is not attached to the object and later gets disabled.  I did not have time to trace why this was occuring.

As a previous poster commented, I am trying to mill multiple objects out of a single piece of plywood.  This bug seems to prevent me from putting tags along particular sides of particular objects.  Things that I have tried to work around the bug (other than hacking at the code) include:
1) Breaking the profiles up so that the milling paths for any particular operation do not overlap at all with other milling paths.
2) Changing the angle of the tag to be 90.
3) Saves/Restarts/Views from different angles and at different degrees of zoom.

tarman3, 2026-01-03:
For `Profile` operation you can try option `SplitArcs`
This will remove some errors

luzpaz, 2026-01-04:
> Version: 1.0.2.39319 (Git) Conda

Can you run 1.2.0-dev and 1.1RC ? https://github.com/FreeCAD/FreeCAD/releases

cdsdonham, 2026-01-05:
I've been trying to figure out how to cause the bug.  My latest thought was considering precision.  As objects get larger, FreeCad changes the US units from inches to feet to yards.  This made me wonder if having objects further from the origin would cause different behavior.  I don't know the units on the "0.1" buried in the version of the Tags code I was looking at.  

Attached is a simple drawing of an 0.5" plywood panel (48" x 96").  At the farthest extent from the origin, I put 4 shapes of random sizes and profiled each one.  On the profile, I then tried to put tags on each vertical or horizontal cut.  The attached drawing shows the tags I was able to put in.  If it doesn't have a tag on a cut, it is because the software would not allow me to put it in.

To see the problem, go into the Job->Operations, and click on DressupTag.  Then click on Add and attempt to add a new tag to any existing line that does not have a tag.

I have tried this file with the code versions shown below.  The behavior is the same with all of them (no ability to put tags on cuts that do not currently show tags).

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
```

And:

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/164.765/1
```

And:

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251230 (Git shallow) AppImage
Build date: 2025/12/30 17:40:39
Build type: Release
Branch: grafted,grafted
Hash: f5759e580b6330906beca12b4be419b030a088ab
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/164.765/1
```

[mwe.zip](https://github.com/user-attachments/files/24425554/mwe.zip)