FreeCAD/FreeCAD #12462: [Problem] Sketcher: No OVP/Widget for Polyline Tool

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Polyline / line set tool currently does not support the use of on-view-parameters (OVP) and makes no use of the widget area in the task dialog.
Polyline could also use the tool widget in the task dialog to display and change the current mode (line, arc, tangent, ...) of the Polyline (M). 
Additionally, the OVP should be displayed (line: angle, length (+maybe toggle for height/width); arc: radius + angle,...)

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-05-18:
Proposed overhaul of PolyLine tool to include OVP

- add quick dimensioning
- optimize it down to 2 modes to cycle between
- for the removed modes, leverage contextual logic to retain functionality
- Address any remaining usability issues from github as priority dictates.
- rename "PolyLine" to "Draw", current line tool left deprecated and accessible via menu or custom toolbar only, flagged for full removal in a future release cycle.

PaddleStroke, 2024-08-30:
Also the overhaul should include fixing : https://github.com/FreeCAD/FreeCAD/issues/13924

PaddleStroke, 2025-03-21:
https://github.com/user-attachments/assets/4936f73b-d10d-4909-b05a-16640c2c4394

Thoughts on this first prototype?

hyarion, 2025-03-21:
@PaddleStroke What is the use case to make a start angle? It looks to me like it is just over-complicating the tool with all these options.
I might not see the case you're trying to solve but maybe it would be enough to support 90 degrees.

If limiting to 90 degree start angles (tangent/perpendicular) then we could simply choose between them how the mouse is moved, similar to how auto-hor/vert constraints are selected. Here it would mean:
1. use the direction between the of starting point and the mouse cursor to decide how to constraint the arc.
2. if user moves the mouse so that the arc length is close to zero, then it rests (goto 1)

Regarding the fillets, can't that be handled automatically if you:
1. create a line
2. add a tangent arc
3. then start a line again. If we don't lock the arc (from step 2) with the ovp, then we could adjust the arc do allow the angle of the line (this one, 3) to change.

I think those changes would allow a better UX.

Another way to solve the fillets would be to allow fillets to be created on selections, that way you would just need to:
1. create polyline like in your example
2. double click on one of the segments of the polyline to select the whole chain
3. click fillet (<-- only this needs to be changed to allow this workflow)

PaddleStroke, 2025-03-21:
- The start angle is a way to support the perpendicular modes without having separated modes. An additional benefit is that it let you use any angle.

I'm not sure what you mean, but I perhaps understood : It would be using the direction of the first movements of the mouse. So if you move the mouse from the starting point along the previous direction, then it starts a tangent arc. If you start in a perpendicular direction then it starts a perpendicular arc.
Is that what you had in mind?
If so we can have this feature on top of the OVP. Keeping the OVP would offer a way to correct things. And set specific angles if necessary.

- For fillet you can do as you suggest. But the advantage of having this fillet options are : 
- - Less edges to draw. You don't have to switch back and forth between line and arc. And you don't have to draw the arcs.
- - The dimensions you enter would include the rounded corners.

Your second idea, of applying fillet to a whole wire is interesting too and could be a feature request. But in this case it is not ideal if you don't want to fillet every corners.

Mannshoch, 2025-03-21:
May as an Idea for other ways to create a tangent or perpendicular tangent (relative to the drawn line):
https://youtu.be/KR8AGCmKKbQ?feature=shared&t=35 
(35s / 53s / 65s)

PaddleStroke, 2025-03-21:
Oh yes that is exactly what I had in mind in my previous message! :)
I should have checked SW polyline

Mannshoch, 2025-03-21:
May as an Idea for filet. With the possibility to show dimension while drawing, why not show radius dimension with value 0 and only create a filet if the user fill it with a different value?

hyarion, 2025-03-21:
> May as an Idea for other ways to create a tangent or perpendicular tangent (relative to the drawn line): https://youtu.be/KR8AGCmKKbQ?feature=shared&t=35 (35s / 53s / 65s)

Dang-it they sole my idea! 😅

I meant exactly that :)

hyarion, 2025-03-21:
@PaddleStroke regarding the "press R to undo" would it be possible to allow normal undo (ctrl+z) to work with it instead?

PaddleStroke, 2025-03-21:
https://github.com/user-attachments/assets/32ca32a9-a962-40fd-9ea9-0608c1ff0cb0
That's working great.

> @PaddleStroke regarding the "press R to undo" would it be possible to allow normal undo (ctrl+z) to work with it instead?

It's a bit tricky actually. @kadet1090 mentioned he is looking into it, if I recall correctly.

hyarion, 2025-03-21:
> That's working great.

👏🤩

> > [@PaddleStroke](https://github.com/PaddleStroke) regarding the "press R to undo" would it be possible to allow normal undo (ctrl+z) to work with it instead?
> 
> It's a bit tricky actually. [@kadet1090](https://github.com/kadet1090) mentioned he is looking into it, if I recall correctly.

Check! :)

maxwxyz, 2025-03-21:
@FreeCAD/design-working-group FYI

Mannshoch, 2025-03-25:
@PaddleStroke  What's your next step? I'm curious what the result will be. (May create a thread on Discord?)

PaddleStroke, 2025-03-25:
Ah yes I got down the rabbit hole of fixing https://github.com/FreeCAD/FreeCAD/pull/20453 so that derailed me a bit from this. It's moving along nicely though, it's mostly working now. I need to do lot of testing and check the 'fillet' idea and see if it can be implemented and would be useful.

PaddleStroke, 2025-03-28:
@AjinkyaDahale I'm also going to look into SketchObject::Fillet function and will likely refactor it

PaddleStroke, 2025-03-28:
@AjinkyaDahale I couldn't find changes to fillet in that PR. Perhaps it was another PR?

AjinkyaDahale, 2025-03-30:
It is in the more generically named commits. They should be in these, but a git blame may give a better idea (you can fetch the branch locally and do it in IDE if github fails because of the sheer size of the file).

```
7421d53392 [Sketcher][DO NOT MERGE] Refactor `SketchObject` some more
b52f24f709 [Sketcher][DO NOT MERGE] Refactor based on loops and extraction into functions
97c6366d35 [Sketcher][DO NOT MERGE] Refactor some loops for understandability
```