FreeCAD/FreeCAD #12464: [Problem] Sketcher: No OVP/Widget for Fillet Tool

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The fillet tool currently does not work with a preselection and makes no use of the OVP (on-view-parameters) and the widget in the task dialog.
It should display the mode in the widget, e.g. simple fillet, corner preserving fillet, and if available a chamfer tool. Cycle though them with the `M`key should be possible.
The OVP should be visible to define the dimension(s) and create the constraints.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
I had this ready in my old branch. I need to port to new architecture

closed 2024-03-18 by WandererFan

MiqCG, 2024-06-12:
Dear, @maxwxyz ,

I found this closed issue after finding #11290 because I think the lack of OVP or "dimensioning on the go" for Fillets and Chamfers in Sketcher is very interesting (for speeding workflow purposes). 

I suppose I don't understand correctly the phases of this issue but I think it has only solved the Widget part, as with the last Appimage (revision 37669), Chamfers and Fillets in Sketcher don't have dimensioning on the go.

If needed I can open a new issue with this feature request (or is this a bug), but I'm afraid I am missing something. Can you point me in the right direction?

https://github.com/FreeCAD/FreeCAD/assets/18265355/a7702d0f-e6c0-4293-ac1f-2ce5d6bea373

PaddleStroke, 2024-06-13:
Indeed there is no OVP for radius. The problem being that it would need to add one click to set the radius so that the arc follows the mouse. As it is in slot for example.
So this needs quite a refactor of things. I had not the time to do it so I've done only the widget.
Feel free to re-open or create a new issue.

PR #12898 (2024-03-18): Sketcher: Chamfer tool (and fillet refactor)
- Rework fillet tool, so that preserve corner is a checkbox not a separate tool.
- Introduce Chamfer tool.
- Few syntax fixes/modernization in fillet()
- unrelated but fix error in className of CmdSketcherCompHorizontalVertical
[image]
Fixes #12464 (though there is no OVP introduced)
Fixes #5623