FreeCAD/FreeCAD #12465: [Problem] Sketcher: Dimension Tool - current mode unclear

maxwxyz, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the smart dimension tool, the current mode is sometimes unclear and also which would be the next mode when cycling through it with the `M` key.
It should make use of the tool widget in the task dialog, to display the current mode and choose the available constraints for the current selection.
When a dimension can be entered, the OVP (on-view-parameters) should be active to quickly enter the dimension without a new popup window opening.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-16:
Can there be current mode for this tool? The initial result depends on the selected object(s) as far as I understand.

maxwxyz, 2024-02-17:
Yes but you can e.g. select two parallel lines, if you have the tool active and press M it switches from dimension to the equal constraint. Or if you select a circle you can cycle through radius or dimension. This feature of cycling does not seem so prominent so a drop down with the current mode and the selection of another mode would be helpful.