FreeCAD/FreeCAD #12466: [Problem] Sketcher_CreateArcOfHyperbola: problem if 1st 2 points have same X coord

Roy-043, 2024-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is not possible to create an arc of hyperbola in the GUI if the first two points have the exact same X coordinate.

Test:
1. Snap to the sketcher grid and click 2 points that have the same X coord.
2. Move the cursor to the right or left, and then up and down.
3. The preview is not usable. It flickers between a vertical line and an almost horizontal curve. Curve endpoint cannot be selected.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=85224

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35858 (Git)
Build type: Release
Branch: main
Hash: 3a73f916ddab6a0af75265acef7cc87e44e4664a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2024-02-17:
See my pull request:  https://github.com/FreeCAD/FreeCAD/pull/12467

luzpaz, 2024-03-15:
@edwilliams16 are you saying that PR impacts this ticket ?

edwilliams16, 2024-03-15:
Yes. It fixes it.

luzpaz, 2024-03-17:
Thanks @edwilliams16!  
@Roy-043  please re-open if you can recreate.  
Resolved by #12467

closed 2024-03-17 by luzpaz