FreeCAD/FreeCAD #12470: [Problem] No way to move or reset the origin of a body/part

maxwxyz, 2024-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD offers no tool or possibility to move/translate or rotate the origin of a part/body (or vice versa, move the body-geometry relative to the objects origin).
This is also a problem with imported geometry (e.g. STEP files), where the origin is way out of the body.
In addition to move the origin to a position or place in precision it should also be possible to reset it to the center of mass of the part/body.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35966 (Git)
Build type: Release
Branch: main
Hash: 7f5d89fa1942fec79222e4d173655744037164dc
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * Defeaturing 1.2.2
  * fasteners 0.5.2
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NomAnor, 2024-02-17:
For STEP files this would be quite handy. This must also handle nested Parts from multi-part STEP files.

For PartDesing Body thats an odd operation, because it would mean to change all attachment offsets of every feature/sketch in the Body. That could be difficult to get right. It's best to design the features in the correct locations from the start.

The only easy way would be to have an explicit "move" feature that transforms the shape of its base feature. That might be useful for the center of mass when you have a complicated shape where its difficult to calculate the center of mass before hand and the feature does the calculation for you.

Ult1Gamer, 2024-05-11:
It would be great to have this implemented!
My application is for creating, or importing 3D files into FreeCAD for use in KiCad.
I would like to see a workflow where I can very easily set the origin of the part/component at the intersection of the plane where the PCB will be relative to that component, and perpendicularly below where the pick-and-place machine will logically pick up the component.
I would be content with a workflow where I create a sketch on the plane that represents where the PCB will touch the component, to place a point, and then set that point as the origin of the component, and and thereafter translating it such that it is in the correct orientation for no adjustments to be necessary in KiCad.

mwganson, 2024-10-19:
Have a look at the TugBoat object in MeshRemodel addon workbench.  Pilot the tug out to the object, latch on to it, and haul it into the harbor.  After you get there have the crew reset the vessel's GPS system so it recognizes the harbor as its home port.

- Select the imported step object.
- Create the TugBoat (this configures the object to the TugBoat's Vessel property, which can also be done in the property editor or from the context menu)
- If the object is in a container, such as in an App::Part, then put the TugBoat into the container, too, from the context menu -> Put into container (menu only shows if there is container)
- Attach the TugBoat to the object.  For the one pictured above, one of the circle edges would be ideal.  How the vessel is positioned relative to the tug will be how it gets positioned relative to the destination.  Internally, the tug boat moves itself to the destination and applies the same transformation to the vessel.  You can't leave the tug at the origin and expect it to haul the vessel to the origin any more than you leave the tug in the harbor and expect it to somehow haul a vessel back in without ever going out to meet it.
- Double click the TugBoat to haul the vessel to the origin.  Optionally, you can pick another destination in the context menu or in properties.  Another TugBoat can serve as the destination, or any object with a Placement will also work.  In the context menu, only another Tug, the local origin, the global origin, or a Part Design LCS, are shown as options, but in the property view anything with a Placement property will work.
- Once at the origin, in the context menu you can use the Set vessel to identity placement or Set vessel to TugBoat placement to reset the object's placement property without moving it.  Option is only available for objects that are Part::Features (not derived from Part::Feature -- so we're talking mostly about imported steps and part::simple copy), Mesh::Features, or Points::Features.  Resetting a parametric object's (example: Part::Box) placement like this would not work because it will simply rebuild its shape on recompute.

FlachyJoe, 2024-10-19:
If we could define a Part children origin it could help.

maxwxyz, 2025-01-19:
Would this be the exposed origin of a body, similar as a tip is for a feature? So all features/sketches reference the true (non editable) origin, and then we have an exposed origin to the outside which can be transformed in relation to the true origin?
@kadet1090 would this work?

kadet1090, 2025-01-19:
I feel that this would cause a lot of potential problems due to mismatch between what is presented to the user and the actual state. I would not want to introduce such layer of indirection  to be honest.

maxwxyz, 2025-01-20:
@kadet1090 but solving https://github.com/FreeCAD/FreeCAD/issues/16106 could close this issue here as well IMO.
If you add a parametric transform feature, you could add it to any body and transfom the shape inside of it, without affecting the origin. Then you'd have a workaround for changing the origin relative to the geometry, right?