FreeCAD/FreeCAD #12473: [Problem] Cancelling Sketcher B-spline creation tools leaves behind construction geometry

Roy-043, 2024-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If any of the 4 Sketcher B-spline creation tools is cancelled by starting another tool after points have been picked, contruction geometry is left behind.

1. Start Sketcher_CreatePeriodicBSplineByInterpolation.
2. Pick several points.
3. Start Sketcher_CreateLine.
4. Only one point from the spline is visible.
5. Pick two points for the line.
6. The other points of the spline also become visible.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-17:
can confirm.
@AjinkyaDahale

wwmayer, 2024-03-16:
It looks like there is a mess with the transactions when creating the spline. After clicking the first point a transaction "Add Knot Point" is added to the Undo list. When finishing it then this transaction is committed.

The behaviour of most other commands is different: they add a transaction after finishing the creation of the geometry.

luzpaz, 2024-10-26:
@AjinkyaDahale is there an ETA on this ?

AjinkyaDahale, 2024-10-26:
> @AjinkyaDahale is there an ETA on this ?

@luzpaz I'm not working on this right now. Do we need this for 1.0?

luzpaz, 2024-10-26:
Not sure. I was wondering if this issue may lead to a memory leak?

AjinkyaDahale, 2024-10-26:
> I was wondering if this issue may lead to a memory leak?

I would be surprised. We can already add all the circles we want by console without it appearing on undo/redo.

zayats1, 2025-07-07:
Weakly 2025/06/30 can confirm (only if press close button, not CTRL + Z)

[image]

[image]

AjinkyaDahale, 2025-07-20:
Likely not a regression. I suspect this change will need quite some overhaul.

PaddleStroke, 2026-09-07:
This should be easy to fix. When the handler is dismiss we just need to make sure the transaction is aborted

closed 2026-09-08 by maxwxyz