FreeCAD/FreeCAD #12479: [Problem] Default shortcut for Sketcher_Dimension (D) conflicts with D key use in Sketcher_CreateBSpline

Roy-043, 2024-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The default shortcut for Sketcher_Dimension (D) conflicts with D key use in Sketcher_CreateBSpline.

1. Start Sketcher_CreateBSpline
2. Click some points.
3. Press D.
4. Result: Sketcher_CreateBSpline is cancelled and Sketcher_Dimension is started.

1. Assign a different shortcut to Sketcher_Dimension.
2. Repeat the previous steps.
3. Result: Spline degree can be specified in a dialog. I am not sure if that actually work BTW.

Documentation:
https://wiki.freecad.org/index.php?title=Sketcher_CreateBSpline&oldid=1364661

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-18:
@PaddleStroke

adrianinsaval, 2024-02-19:
dimension seems like a much more frequent usecase so it should probably keep the simplest shortcut, or it could be K, D like other constraints.
I'm not sure what shortcut should eb used instead in bspline

PaddleStroke, 2024-02-20:
> What does DSH stand for?

DrawSketchHandler. It's the class that contains the logic of sketcher
geometry tools.

adrianinsaval, 2024-05-07:
related #12016

closed 2024-06-24 by chennes