FreeCAD/FreeCAD #12480: [Problem] Sketcher_CreateBSplineByInterpolation Unhandled unknown C++ exception if grid snap is on

Roy-043, 2024-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Enable grid snap.
2. Call Sketcher_CreateBSplineByInterpolation or Sketcher_CreatePeriodicBSplineByInterpolation.
3. Pick a point, no need to actually snap.
4. Move the cursor.
5. Result: Unhandled unknown C++ exception

Full error message:
```
21:36:15  Unhandled unknown C++ exception in ViewProvider::eventCallback (Event type: SoLocation2Event, object type: SketcherGui::ViewProviderSketch)
```

### Full version info

OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35994 (Git)
Build type: Release
Branch: main
Hash: 69097667df47b2cc86d8688d2dbb545319e33e68
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-16:
> Enable grid snap.

Where and how? I cannot find this option any more.

wwmayer, 2024-03-16:
https://github.com/FreeCAD/FreeCAD/pull/12999

commit fc8f93d4 (2024-03-16): Sketch: fixes #12480: Sketcher_CreateBSplineByInterpolation Unhandled unknown C++ exception

The interpolation of a BSpline can fail and then OCC throws an exception. Because this OCC exception isn't handled by the DrawSketchHandler the core system handles it as unknwon C++ exception
closed 2024-03-16 by wwmayer